moveit/moveit2 #373: Remove Cartesian Interpolator and point documentation to use OMPL constrained planner

nbbrooks, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
Now that the [OMPL constrained planner](https://github.com/ros-planning/moveit2/pull/347) is merged in, we should remove the Cartesian Interpolator code and any tutorial references. The default method for Cartesian planning should be the OMPL planner. The cartesian interpolator has poor performance and this will improve MoveIt's maintainability by reducing the size of the codebase and feature set.

**Describe the solution you'd like**
- [ ] Delete Cartesian Interpolator code
- [ ] Update cartesian planning tutorials to reference OMPL planner

felixvd, 2021-02-25:
I'm not sure I agree. The interpolator is simple and fast, which is an advantage (I remember this coming up in another issue on the moveit repo). And even if we decided to cut it, wouldn't the pilz_industrial_motion_planner's LIN motion be a more appropriate replacement?

davetcoleman, 2021-03-04:
We would need to get the alternative Cartesian planner (OMPL, Pilz, or whatever) to be similarly easy to setup and use. But I can tell you that one of the #1 issues we fix for our clients at PickNik are the limitations of the Cartesian Interpolator. It really feels wrong to me how poorly it addresses joint limits, singularities, etc yet is promoted in the tutorials.