moveit/moveit2 #381: Issue when overriding StateValidityChecker methods

thibaultbarbie, 2021-03-11:
### Description

I am trying to integrate a new planner in OMPL and use it in Moveit. I built OMPL from source and by using make install updated the libompl that (I think) Moveit links to. I am able to use my planner in Moveit and my planner is also able to use the StateValidityChecker to verify if a state is valid or not.

However, for my planner I would like to have some more information (distance of points in joints to the closest obstacle) and for that I wanted to extend the StateValidityChecker class (in OMPL). This class is used as parent in Moveit with a class of same name [StateValidityChecker](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h)  and two methods (isValid and clearance) are overriding the virtual methods in the OMPL class [StateValidityChecker](https://github.com/ompl/ompl/blob/master/src/ompl/base/StateValidityChecker.h).

My problem is that if I create a new method in the OMPL class StateValidityChecker and override it in the Moveit class StateValidityChecker the compiler complains that the method is marked override but does not override. 

Possibly related: To understand better my problem I read the Moveit source code and I am utterly confused on this line https://github.com/ros-planning/moveit2/blob/2a10dc79a284e29edfeb05c156f3eb6fb0030d2d/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L136
I am not that skilled in c++ but my understanding is that the ModelPlanningContext instance is cast as a StateValidityChecker, how is it possible? It does not possess the right methods.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build version 2.1.0

### Steps to reproduce
1. Add a new method in the StateValidityChecker class in OMPL.
2. Override it in the StateValidityChecker class in Moveit.

### Expected behaviour
It should compile and the method implemented in the Moveit class should be performed.

### Actual behaviour
It does not compile. 

### Backtrace or Console output

https://gist.github.com/thibaultbarbie/bbb9e9d27807a7dd3609e78172b1efa2

JafarAbdi, 2021-03-11:
Hi @thibaultbarbie,

This's not a bug in MoveIt, from the error message you're trying to override a non-existing virtual function `my_method` which is expected to fail

If you can share your code I'm super happy to help you get it to work

closed 2021-03-11 by JafarAbdi

thibaultbarbie, 2021-03-16:
Thank you for your help @JafarAbdi !

I tried many things and discover that it should be a linking problem. In my previous message I showed an overriding problem but in reality it is more general. Anything that I add in the ompl source code outside of the previously defined methods or classes seem to not be recognized by moveit. It is like the "interface" is frozen but I can modify the content of the already defined methods.

Also, after tinkering with it I discovered that moveit will load the ompl shared library from /opt/ros/foxy/lib/libompl.so.

So, for a minimal working example of my problem:
1. Download the source code of ompl.
2. Change ${CMAKE_INSTALL_LIBDIR} to "/opt/ros/foxy/lib" in src/ompl/CMakeList.txt
3. Modify the source by adding a new method or class. For example add a new method in ompl/src/ompl/base/StateValidityChecker.h
4. Compile: make && make install
5. Modify the moveit2 source code to use the added method or class. For example in moveit2/moveit_planners/ompl/ompl_interface/src/detail/state_validity_checker.cpp
6. Compile moveit2

It should throw an error saying that the method/class you want to use does not exist.

However, the weird thing is that if you modify something in the current method/class then it works. I am fairly inexperienced with shared libraries but it does seem that the interface is fixed.

JafarAbdi, 2021-03-16:
@thibaultbarbie Do you mind sharing your example that is failing currently .?

thibaultbarbie, 2021-03-18:
I found the problem! I was copying the shared library .so files to /opt/ros/foxy/lib but I was not copying the headers .h to the include folder. That is why the interface was "frozen" and I could not add any methods or variables. Thank you!