moveit/moveit2 #384: moveit_cpp and move_group_interface demos crash

AndyZe, 2021-03-15:
Following the "build from source" instructions here, this demo crashes:

`ros2 launch run_moveit_cpp run_moveit_cpp.launch.py`

> Task exception was never retrieved
> future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=SubstitutionFailure("package 'run_moveit_cpp' found at '/home/andy/ws_servo2/install/run_moveit_cpp', but libexec directory '/home/andy/ws_servo2/install/run_moveit_cpp/lib/run_moveit_cpp' does not exist")>

This runs well:

`ros2 launch run_move_group run_move_group.launch.py`

but this crashes:

`ros2 launch run_move_group run_move_group_interface.launch.py`

`package 'run_move_group' found at '/home/andy/ws_servo2/install/run_move_group', but libexec directory '/home/andy/ws_servo2/install/run_move_group/lib/run_move_group' does not exist`

The Servo demo seems to be fine.

AndyZe, 2021-03-15:
I triple-checked that I was sourced correctly.

liborw, 2021-03-15:
Hi I had simmilar problem, my setup is that I have moveti2 workspace and my workspace and for experiments with this demo i have copied just the run_move_group package to my workspace. When run from the moveit2 workspace it works fine when from my workspace it crashes with that error. 

With help of google I found out that the package is compiled only inside the "global" path so you can run `run_move_group` alone without `ros2 run`, but the `ros2 launch` expect it to be at the `.../lib/run_move_group`. This is what I had to change in order to make it install into `lib` but I'm not sure whether everything is needed or whether it does not break anything else:

```
diff --git a/moveit_demo_nodes/run_move_group/CMakeLists.txt b/moveit_demo_nodes/run_move_group/CMakeLists.txt
index c9093cce7..4a505e212 100644
--- a/moveit_demo_nodes/run_move_group/CMakeLists.txt
+++ b/moveit_demo_nodes/run_move_group/CMakeLists.txt
@@ -1,10 +1,6 @@
 cmake_minimum_required(VERSION 3.5)
 project(run_move_group)
 
-# Common cmake code applied to all moveit packages
-find_package(moveit_common REQUIRED)
-moveit_package()
-
 if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
   add_compile_options(-Wpedantic)
 endif()
@@ -26,13 +22,8 @@ ament_target_dependencies(run_move_group
 )
 
 install(TARGETS run_move_group
-  EXPORT export_${PROJECT_NAME}
-  LIBRARY DESTINATION lib
-  ARCHIVE DESTINATION lib
-  RUNTIME DESTINATION bin
-  INCLUDES DESTINATION include
+  DESTINATION lib/${PROJECT_NAME}
 )
-ament_export_targets(export_${PROJECT_NAME})
 
 install(DIRECTORY launch
   DESTINATION share/${PROJECT_NAME}
```

PR #418 (2021-04-13): Fix node install directory in run_move_group
Fixes #384 and #415. 

run_moveit_cpp was already fixed by a previous commit.

run_move_group_interface now correctly launches but does nothing? I am not sure what it supposed to do. @JafarAbdi can you check it
closed 2021-04-13 by JafarAbdi