moveit/moveit2 #387: Fail to build from source

alan0428a, 2021-03-18:
### Description

I'm following the [instructions ](https://moveit.ros.org/install-moveit2/source/) to build moveit2. Everything works fine until I run the build script
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

It shows that some variables are not set correctly, below is the detailed information. Is there anything I missed?

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: Ubuntu 20.04
* Source build
* If source, which branch?
Following the setting in https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos

### Steps to reproduce
Following the instruction.

### Expected behaviour
Build correctly.

### Actual behaviour
Fail to build.

### Backtrace or Console output

[image]

swilcock0, 2021-03-19:
@alan0428a To me this looks like you may not have Boost installed, could you try running `sudo apt-get install libboost-all-dev` first

alan0428a, 2021-03-19:
> @alan0428a To me this looks like you may not have Boost installed, could you try running `sudo apt-get install libboost-all-dev` first

Thanks for the reply. However, it seems like I've already installed the lib.
```Reading package lists... Done
Building dependency tree       
Reading state information... Done
libboost-all-dev is already the newest version (1.71.0.0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

JafarAbdi, 2021-04-15:
You should have a `/usr/share/cmake-**VERSION**/Modules/FindBoost.cmake` file that will find the boost components for you, could you please check if it exists .?

JafarAbdi, 2021-04-15:
I'll close this, please feel free to reopen it

closed 2021-04-15 by JafarAbdi