moveit/moveit2 #388: Request for moveit_setup_assistant support

JaehyunShim, 2021-03-18:
**Is your feature request related to a problem? Please describe.**
It seems the MoveIt dev team hasn't started working on the future moveit_setup_assistant support yet. I would like to ask about the plan for it. If the team is too busy to work on it, I am interested in contributing to it since I am going to work on something similar to it (building a setup assistant) in the future.

Thanks, 
Jaehyun

henningkayser, 2021-05-31:
We've started planning for future MSA support (see #313 and attached milestone issues). Because of the fundamental changes to the ROS launch infrastructure, we decided to do a redesign of the MoveIt config package and with that a refactoring of the MSA. A complete straight port doesn't make too much sense, unfortunately, but you are very welcome to add to the discussion or contribute. If you are still willing and have capacity to put work into MSA, please let me know and we can sync up to coordinate this effort.

JaehyunShim, 2021-05-31:
THanks! I will close this issue and will look forward to the new setup assistant.

closed 2021-05-31 by JaehyunShim