moveit/moveit2 #397: Flaky tests in moveit_servo

tylerjw, 2021-03-25:
### Description

When migrating to GitHub Actions I discovered that two of the moveit_servo tests are flaky.  In that PR I disabled them by commenting out them from the CMakeLists.txt.  This issue is to track fixing those tests.  I'm marking this good-first-issue because it is an isolated task.  However, teasing out why they are flaky might not be trivial.

The tests are:
* unit_test_servo_calcs
* test_servo_pose_tracking

NickNair, 2021-04-17:
On launching which moveit_servo nodes are these called ?

NickNair, 2021-04-20:
the Description for unit_test_servo_calcs says - Loads a Servo instance without launching or parameters. Can I get some more description ?

PR #603 (2021-08-16): Enable Servo Tests
With the recent fixes on Rolling and Galactic's `ComposobleNodeContainer`, we should be able to enable these tests safely. Closes #397.
closed 2021-08-16 by henningkayser