moveit/moveit2 #399: move_group_ompl_constraints_test is failing on main

tylerjw, 2021-03-26:
### Description

As seen in [this test run on main](https://github.com/ros-planning/moveit2/runs/2202814511) `move_group_ompl_constraints_test` is now failing.

@JafarAbdi you helped me get this to pass before, do you have any ideas what is happening here... to me it looks like it's a couple of the numbers in the last column of the transform matrix that are different.

```    [move_group_ompl_constraints_test-5] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/planning_interface/test/move_group_ompl_constraints_test.cpp:100: planAndMove
    [move_group_ompl_constraints_test-5] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/planning_interface/test/move_group_ompl_constraints_test.cpp:136: PositionConstraint
    [move_group_ompl_constraints_test-5] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/planning_interface/test/move_group_ompl_constraints_test.cpp:122: Failure
    [move_group_ompl_constraints_test-5] Value of: actual_pose.isApprox(goal_pose, EPSILON)
    [move_group_ompl_constraints_test-5]   Actual: false
    [move_group_ompl_constraints_test-5] Expected: true
    [move_group_ompl_constraints_test-5] expected: 
    [move_group_ompl_constraints_test-5]     0.707388    -0.706825  4.62412e-17      0.30702
    [move_group_ompl_constraints_test-5]    -0.706825    -0.707388 -6.92765e-12          0.3
    [move_group_ompl_constraints_test-5]  4.89667e-12  4.90051e-12           -1      0.29027
    [move_group_ompl_constraints_test-5]            0            0            0            1
    [move_group_ompl_constraints_test-5] actual: 
    [move_group_ompl_constraints_test-5]     0.707388    -0.706825  4.62412e-17      0.30702
    [move_group_ompl_constraints_test-5]    -0.706825    -0.707388 -6.92765e-12 -5.22133e-12
    [move_group_ompl_constraints_test-5]  4.89667e-12  4.90051e-12           -1      0.59027
    [move_group_ompl_constraints_test-5]            0            0            0            1
    [move_group_ompl_constraints_test-5] Google Test trace:
    [move_group_ompl_constraints_test-5] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/planning_interface/test/move_group_ompl_constraints_test.cpp:112: testPose
    [move_group_ompl_constraints_test-5] /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/planning_interface/test/move_group_ompl_constraints_test.cpp:136: PositionConstraint
    [move_group_ompl_constraints_test-5] [  FAILED  ] ConstrainedPlanningTestFixture.PositionConstraint (6616 ms)
    [move_group_ompl_constraints_test-5] [----------] 1 test from ConstrainedPlanningTestFixture (6617 ms total)
    [move_group_ompl_constraints_test-5] 
    [move_group_ompl_constraints_test-5] [----------] Global test environment tear-down
    [move_group_ompl_constraints_test-5] [==========] 1 test from 1 test case ran. (6621 ms total)
    [move_group_ompl_constraints_test-5] [  PASSED  ] 0 tests.
    [move_group_ompl_constraints_test-5] [  FAILED  ] 1 test, listed below:
    [move_group_ompl_constraints_test-5] [  FAILED  ] ConstrainedPlanningTestFixture.PositionConstraint
    [move_group_ompl_constraints_test-5] 
    [move_group_ompl_constraints_test-5]  1 FAILED TEST
Error: ROR] [move_group_ompl_constraints_test-5]: process has died [pid 21954, exit code 1, cmd '/home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_ros_planning_interface/move_group_ompl_constraints_test --ros-args --params-file /tmp/launch_params_m8ll4gbd --params-file /tmp/launch_params_madt4g10 --params-file /tmp/launch_params__54usvpq'].
```

swilcock0, 2021-03-29:
Unsure about the why as I'm still figuring out how ros2_control works, but looking at the test and the log, this actual position indicates that the move() was completely unsuccessful - and earlier in the log it shows that the controller fails to start due to the switch_controller service:

>  [ros2 control switch_controllers --start-controllers panda_arm_controller-8] Could not contact service /controller_manager/switch_controller
Error: ROR] [ros2 control switch_controllers --start-controllers panda_arm_controller-8]: process has died [pid 22046, exit code 1, cmd 'ros2 control switch_controllers --start-controllers panda_arm_controller'].

JafarAbdi, 2021-03-29:
I think there are two causes:

1- Currently we're loading the controllers by using [ExecuteProcess](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/launch/move_group_launch_test_common.py#L168) and there's no guarantee that this process will be called after [ros2_control_node](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/launch/move_group_launch_test_common.py#L149) being launched, so sometimes as @swilcock0 mentioned it will fail to contact the service

2- We still don't have support for multi-dof joints in the current state monitor causing the `getCurrentState` function to always wait until it timeout which causes the trajectory execution manager to take more time and it timeout before it finishes execution

I'm working on fixing both this week

tylerjw, 2021-03-30:
> 2- We still don't have support for multi-dof joints in the current state monitor causing the `waitForCurrentState` function to always wait until it timeout which causes the trajectory execution manager to take more time and it timeout before it finishes execution

I wonder if this is one of the main reasons for the flakyness in servo now too.

closed 2021-04-01 by tylerjw