moveit/moveit2 #400: Source Build instructions never clone the moveit2 repo

AndyZe, 2021-03-28:
If you follow the instructions [here](https://moveit.ros.org/install-moveit2/source/) carefully, the moveit2 repo is not cloned. It is not part of [moveit2.repos](https://github.com/ros-planning/moveit2/blob/main/moveit2.repos).

We should probably add moveit2 to moveit2.repos.

tylerjw, 2021-03-28:
I created this https://github.com/ros-planning/moveit.ros.org/pull/577 when I removed it from moveit2.repos.  I'm now using that repos file as part of the CI system to define upstream dependencies.  Having moveit2 as a dependency of moveit2 doesn't work.  This also makes sure you can build all the upstream libraries in an underlay and they'll work.

AndyZe, 2021-03-29:
Good job creating a patch right away! I guess there was a regrettable delay in getting it merged.

closed 2021-03-29 by AndyZe

tylerjw, 2021-03-29:
Thank you for merging it... it was tiny and I should have asked someone, I just forgot.