moveit/moveit2 #411: geometric_shapes compile failed and can't load Robot model

chaos4ros2, 2021-04-06:
### Description

I'm following the instructions to build moveit2 but 1 package failed.（geometric_shapes）
The same problem issue found in the [link](https://github.com/qhull/qhull/issues/78) below but still not fixed.

Ignore it and try to ran MoveItCpp Demo Package by use command 「_ros2 launch run_moveit_cpp run_moveit_cpp.launch.py_」but the RViz GUI didn't launched.

Than I use another terminal to launch rviz by use command「_rviz2 -d ~/ws_moveit2/src/moveit2/moveit_demo_nodes/run_moveit_cpp/launch/run_moveit_cpp.rviz_」,the panda robot's tf flame appeared but urdf model can't show and error is displayed in terminal：
**[INFO] [1617523472.376519700] [moveit_rdf_loader.rdf_loader]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1617523472.520297400] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded**

Is this error related to the failed package and how can I fix it?

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: Ubuntu 20.04 （wsl2 windsow10）
* Source or Binary build? Source
* If source, which branch? follow this [link](https://moveit.ros.org/install-moveit2/source/) and didn't change anything

### Steps to reproduce
compile in windos10 wsl2

### Expected behaviour
compile success and no rviz2's error 

### Actual behaviour
geometric_shapes compile failed and couldn't load Robot model

tylerjw, 2021-04-06:
The error about the robot model not being found has nothing to do with the error output from `geometric_shapes`.  Is there more error output than that?

chaos4ros2, 2021-04-06:
Yes I will show it as detailed as possible.

first,I use  command 「_ros2 launch run_moveit_cpp run_moveit_cpp.launch.py_」
and the message show like below:

_chaos@LAPTOP-CDCTQI88:~/ws_moveit2$ source ~/ws_moveit2/install/setup.bash_
_chaos@LAPTOP-CDCTQI88:~/ws_moveit2$ ros2 launch run_moveit_cpp run_moveit_cpp.launch.py_
**[INFO] [launch]: All log files can be found below /home/chaos/.ros/log/2021-04-06-21-41-50-926321-LAPTOP-CDCTQI88-5800
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [5802]
[INFO] [robot_state_publisher-2]: process started with pid [5804]
[INFO] [rviz2-3]: process started with pid [5806]
[INFO] [run_moveit_cpp-4]: process started with pid [5808]
[INFO] [ros2_control_node-5]: process started with pid [5810]
[INFO] [ros2 run controller_manager spawner.py panda_arm_controller-6]: process started with pid [5812]
[INFO] [ros2 run controller_manager spawner.py joint_state_controller-7]: process started with pid [5814]
[static_transform_publisher-1] [INFO] [1617712911.312092300] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link panda_link1 had 1 children
[robot_state_publisher-2] Link panda_link2 had 1 children
[robot_state_publisher-2] Link panda_link3 had 1 children
[robot_state_publisher-2] Link panda_link4 had 1 children
[robot_state_publisher-2] Link panda_link5 had 1 children
[robot_state_publisher-2] Link panda_link6 had 1 children
[robot_state_publisher-2] Link panda_link7 had 1 children
[robot_state_publisher-2] Link panda_link8 had 1 children
[robot_state_publisher-2] Link panda_hand had 2 children
[robot_state_publisher-2] Link panda_leftfinger had 0 children
[robot_state_publisher-2] Link panda_rightfinger had 0 children
[robot_state_publisher-2] [INFO] [1617712911.323456400] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-2] [INFO] [1617712911.323614800] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-2] [INFO] [1617712911.323641100] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-2] [INFO] [1617712911.323657900] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-2] [INFO] [1617712911.323671900] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-2] [INFO] [1617712911.323685000] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-2] [INFO] [1617712911.323697900] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-2] [INFO] [1617712911.323713500] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-2] [INFO] [1617712911.323730700] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-2] [INFO] [1617712911.323747200] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-2] [INFO] [1617712911.323761900] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-2] [INFO] [1617712911.323775600] [robot_state_publisher]: got segment panda_rightfinger
[run_moveit_cpp-4] [INFO] [1617712911.339648900] [moveit_cpp_demo]: Initialize node
[ros2_control_node-5] [INFO] [1617712911.343830900] [controller_manager]: update rate is 100 Hz
[rviz2-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-chaos'
[rviz2-3] [INFO] [1617712911.716335800] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1617712911.716670300] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-3] [INFO] [1617712911.766087500] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [INFO] [1617712911.834757000] [controller_manager]: Loading controller 'joint_state_controller'
[ros2 run controller_manager spawner.py panda_arm_controller-6] [INFO] [1617712911.838809300] [spawner_panda_arm_controller]: Waiting for /controller_manager services
[ros2 run controller_manager spawner.py joint_state_controller-7] [INFO] [1617712911.851489700] [spawner_joint_state_controller]: Loaded joint_state_controller
[ros2_control_node-5] [INFO] [1617712911.854822500] [controller_manager]: Configuring controller 'joint_state_controller'
[ros2 run controller_manager spawner.py joint_state_controller-7] [INFO] [1617712911.871518200] [spawner_joint_state_controller]: Configured and started joint_state_controller
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] [INFO] [1617712911.922562500] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.001968 seconds
[rviz2-3] [INFO] [1617712911.922770800] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[INFO] [ros2 run controller_manager spawner.py joint_state_controller-7]: process has finished cleanly [pid 5814]
[ros2_control_node-5] [INFO] [1617712912.044993900] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2 run controller_manager spawner.py panda_arm_controller-6] [INFO] [1617712912.064005000] [spawner_panda_arm_controller]: Loaded panda_arm_controller
[ros2_control_node-5] [INFO] [1617712912.066913500] [controller_manager]: Configuring controller 'panda_arm_controller'
[ros2_control_node-5] [INFO] [1617712912.068224200] [panda_arm_controller]: Controller state will be published at 50Hz.
[ros2_control_node-5] [INFO] [1617712912.069107600] [panda_arm_controller]: Action status changes will be monitored at 20Hz.
[... cut ...]
you can see rviz2 node launched and added in windows task bar. But when I mouse over it, nothing happened just show a small white screen.

Then, I use command 「_rviz2 -d ~/ws_moveit2/src/moveit2/moveit_demo_nodes/run_moveit_cpp/launch/run_moveit_cpp.rviz_」in another teminal and rviz2 node launched twice and nothing happened too but show error message below with red word.
[image]

Finally,I just use command 「_rviz2_」and this time RViz GUI luanched! I add the plugin manually and RobotModel is not loaded too. moveit tool also not works.

[image]

that's all what happened.

tylerjw, 2021-04-06:
Do you mind quoting the code block with three back ticks or using the code button in the wsiwyg edditor to do it so the block of errors is easier to read?

tylerjw, 2021-04-06:
So it seems that there is some issue with launching rviz.  I don't see any errors about rviz which is odd.  Perhaps we are now missing some needed argument.  I don't have a windows machine to test this on, I'm sorry.

The errors you are getting when you launch rviz with that config instead of trying to do it through the demo is that you are just opening rviz with a config file (which specifies things like what topics it should listen to) and you are not starting the nodes.  If you open rviz using that command and then try to run the demo separately does that work?

chaos4ros2, 2021-04-06:
I try two ways

**1.** 
step1: terminal_1 _「ros2 launch run_moveit_cpp run_moveit_cpp.launch.py」_
step2: terminal_2 _「ros2 launch run_moveit_cpp run_moveit_cpp.launch.py」_

result: both rviz2 can't launched.

step1

[image]

step2
[image]

**2.**
step1: terminal_1 _「rviz2」_
step2: terminal_2 _「ros2 launch run_moveit_cpp run_moveit_cpp.launch.py」_
step3: add plugins in rviz2

result: rviz launched but show error

step1
[image]

step2
[image]

chaos4ros2, 2021-04-06:
> So it seems that there is some issue with launching rviz. I don't see any errors about rviz which is odd. Perhaps we are now missing some needed argument. I don't have a windows machine to test this [...]

123

chaos4ros2, 2021-04-06:
> 
> 
> Do you mind quoting the code block with three back ticks or using the code button in the wsiwyg edditor to do it so the block of errors is easier to read?

I am sorry I don't know how to use wsiwyg edditor but I turn code block to 「code」mode and add add line break.
Is it easier to read?

tylerjw, 2021-04-06:
What I meant about the code blocks was to put this before and after a code block:

\`\`\`

tylerjw, 2021-04-06:
@JafarAbdi @henningkayser have either of you seen this (and do you have a windows machine configured to easily try to reproduce this)?  I don't have a windows machine to test on and this does work for me on ubuntu.

chaos4ros2, 2021-04-07:
> 
> 
> What I meant about the code blocks was to put this before and after a code block:
> 
> ```

got it, I will do this next time.

chaos4ros2, 2021-04-07:
> 
> 
> What I meant about the code blocks was to put this before and after a code block:
> 
> ```

Thank you for your advice. I fixed it.

JafarAbdi, 2021-09-28:
@chaos4ros2 This's fixed, right .? closing for now

closed 2021-09-28 by JafarAbdi

chaos4ros2, 2021-09-29:
I apologize for leaving this issue unresolved. I will try it later and reopen it if the problem still exists.