moveit/moveit2 #422: moveit2 setup assistant

vtr101, 2021-04-15:
Hello,

I'm currently trying to simulate a robotic manipulator using ROS2 and I want to use moveit2 to do it. However, I don't know how to convert the moveit1 launch and yaml files to a ROS2 format. Is there a more complete example of the launch files generated by the setup assistant in a ROS2 format?

Thank you.

JafarAbdi, 2021-04-15:
Hi @vtr101,

> Is there a more complete example of the launch files generated by the setup assistant in a ROS2 format?

For examples on how the config & launch files look like in ROS2 please take a look at [config](https://github.com/ros-planning/moveit_resources/tree/ros2/panda_moveit_config/config) & [launch](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group/launch)

JafarAbdi, 2021-04-15:
I'm going to close this issue, please feel free to ask more questions

closed 2021-04-15 by JafarAbdi