moveit/moveit2 #425: moveit2 demo question

chaos4ros2, 2021-04-16:
**Is your feature request related to a problem? Please describe.**
I want to learn how to use MoveIt2 with python and found a tutorial in [move_group_python_interface_tutorial](http://moveit2_tutorials.picknik.ai/doc/move_group_python_interface/move_group_python_interface_tutorial.html)
but the content is still MoveIt1.Then I found a similar demo in [moveit_demo_nodes/run_move_group](https://github.com/ros-planning/moveit2/tree/main/moveit_demo_nodes/run_move_group)
but can't locate a python interface.

**my question**
・Is there any python interface to use MoveIt2 ?  
・if not,is it a good way to create a MoveIt2's python interface based on [move_group_python_interface_tutorial](http://moveit2_tutorials.picknik.ai/doc/move_group_python_interface/move_group_python_interface_tutorial.html)
・the moveit_commander module used in [move_group_python_interface_tutorial.py](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py) but not appeard in MoveIt2 Package、Is there any alternatives in MoveIt2 ? If not, is it a good idea to remake moveit_commander?

swilcock0, 2021-04-16:
Hi @chaos4ros2 , so far the moveit_commander hasn’t been ported over yet - there’s a tracking issue for it here #314 . Looking at ros-planning/moveit#900 it seems that it’s on its way. My personal alternative has been to set up my own services in C++ that I can call from python

chaos4ros2, 2021-04-18:
Thank you for the easy to understand explanation.
Now I know moveit_commander is the module to call MoveGroupInterface(C++) by use boost::python and replace 
boost::python by pybind11, a new version is under development. It's really good news for me.

Before release, I will try to rewrite these two [demos](https://github.com/ros-planning/moveit_commander/tree/260f5724a8f7b906176d5439b8fbf2d9af8f7908/demos) for moveit2. Not like [move_group_python_interface_tutorial.py](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py)
these [demos](https://github.com/ros-planning/moveit_commander/tree/260f5724a8f7b906176d5439b8fbf2d9af8f7908/demos) use less moveit_commander's class so may be easier to rewrite and understand.
I am sorry I am a C++ beginner so it's to hard to use C++ interface directly now and don't know how to call C++ services(nodes?)
form python.

chaos4ros2, 2021-04-23:
I try to run [plan.py](https://github.com/ros-planning/moveit_commander/blob/kinetic-devel/demos/plan.py) so I create a work_space like [this](https://github.com/chaos4ros2/moveit2_commander).
plan.py needs moveit_commander so put it in work_space too.
Then I run _colcon build --symlink-install_、_source install/setup.bash_
and start up lanuch file by command _ros2 launch moveit2_commander demo.launch.py_

the error show below
```
chaos@chaos4ros2:~/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws$ ros2 launch moveit2_commander demo.launch.py
[INFO] [launch]: All log files can be found below /home/chaos/.ros/log/2021-04-23-09-18-42-465832-chaos4ros2-14300
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [plan-1]: process started with pid [14302]
[plan-1] Traceback (most recent call last):
[plan-1]   File "/home/chaos/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws/install/moveit2_commander/lib/moveit2_commander/plan", line 11, in <module>
[plan-1]     load_entry_point('moveit2-commander', 'console_scripts', 'plan')()
[plan-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[plan-1]     return get_distribution(dist).load_entry_point(group, name)
[plan-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[plan-1]     return ep.load()
[plan-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[plan-1]     return self.resolve()
[plan-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[plan-1]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[plan-1]   File "/home/chaos/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws/build/moveit2_commander/moveit2_commander/plan.py", line 11, in <module>
[plan-1]     from moveit_commander import RobotCommander, roscpp_initialize, roscpp_shutdown
[plan-1]   File "/home/chaos/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws/build/moveit_commander/moveit_commander/__init__.py", line 2, in <module>
[plan-1]     from .roscpp_initializer import *
[plan-1]   File "/home/chaos/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws/build/moveit_commander/moveit_commander/roscpp_initializer.py", line 35, in <module>
[plan-1]     from moveit_ros_planning_interface import _moveit_roscpp_initializer
[plan-1] ModuleNotFoundError: No module named 'moveit_ros_planning_interface'
[ERROR] [plan-1]: process has died [pid 14302, exit code 1, cmd '/home/chaos/moveit2_ws/src/moveit2_tutorials/doc/moveit2_commander_ws/install/moveit2_commander/lib/moveit2_commander/plan --ros-args -r __node:=plan'].
```

I installed moveit2 correctly、and 「moveit_ros_planning_interface」is included in the pkg list（confirm it by ros2 pkg list）
I'm not familiar with build system. What else do I have to do?

swilcock0, 2021-04-23:
@chaos4ros2 [roscpp_initializer.py](https://github.com/ros-planning/moveit2/blob/main/moveit_commander/src/moveit_commander/roscpp_initializer.py) is trying to load a module from the moveit_ros/planning interface python module - looking at [moveit_ros/planning_interface/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/CMakeLists.txt) the lines relating to Python have been commented out, so the Python side of it is not being built with Moveit. It _may_ work if you try uncommenting these lines, but I believe they are commented out for a reason! 😄

chaos4ros2, 2021-04-24:
Thank you for the advice and work is progressing but get error again.
I uncomment as you tell me and rebuilt moveit2 package

I try two ways but get error both.

First, I uncomment all package with keyword 「python」in [moveit_ros/planning_interface/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/CMakeLists.txt) 
```
24 find_package(PythonInterp REQUIRED) 
25 find_package(PythonLibs "${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}" REQUIRED)
28 find_package(eigenpy REQUIRED)
34 py_bindings_tools/include
46 moveit_py_bindings_tools
47 moveit_py_bindings_tools_python
70 add_subdirectory(py_bindings_tools)
74 add_subdirectory(robot_interface)
75 add_subdirectory(test)
```
Than rebuilt moveit2 and error show below
```
-- stderr: moveit_ros_planning_interface
CMake Error at CMakeLists.txt:28 (find_package):
  By not providing "Findeigenpy.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "eigenpy", but
  CMake did not find one.

  Could not find a package configuration file provided by "eigenpy" with any
  of the following names:

    eigenpyConfig.cmake
    eigenpy-config.cmake

  Add the installation prefix of "eigenpy" to CMAKE_PREFIX_PATH or set
  "eigenpy_DIR" to a directory containing one of the above files.  If
  "eigenpy" provides a separate development package or SDK, be sure it has
  been installed.
```
I think this error is caused by missing [eigenpy](https://index.ros.org/r/eigenpy/) package.
I want to install [eigenpy](https://index.ros.org/r/eigenpy/) later so I  commented out line 28 and rebuilt again.

Then I get error like this:
```
--- stderr: moveit_ros_planning_interface
CMake Error at py_bindings_tools/CMakeLists.txt:28 (install):
  install DIRECTORY given no DESTINATION!

CMake Error at test/CMakeLists.txt:6 (find_package):
  By not providing "Findeigen_conversions.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_conversions", but CMake did not find one.

  Could not find a package configuration file provided by "eigen_conversions"
  with any of the following names:

    eigen_conversionsConfig.cmake
    eigen_conversions-config.cmake

  Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
  set "eigen_conversions_DIR" to a directory containing one of the above
  files.  If "eigen_conversions" provides a separate development package or
  SDK, be sure it has been installed.

make: *** [Makefile:350: cmake_check_build_system] エラー 1
---
```
It seems some problem occurred in [py_bindings_tools/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/py_bindings_tools/CMakeLists.txt)

I think the reason of error is because of using catkin so I follow this [guide ](https://docs.ros.org/en/foxy/Contributing/Migration-Guide.html#update-the-cmakelists-txt-to-use-ament-cmake)and try to replace variables with 
keyword 「catkin」.

But the question is some variables don't appeared in guide so I don't know how to replace it.
```
CATKIN_GLOBAL_BIN_DESTINATION => bin
CATKIN_GLOBAL_INCLUDE_DESTINATION => include
catkin_LIBRARIES => ?
CATKIN_DEVEL_PREFIX => ?
CATKIN_PACKAGE_PYTHON_DESTINATION => ?
```

chaos4ros2, 2021-04-25:
Progress report：

I solve the problem in [moveit_ros/planning_interface/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/CMakeLists.txt) while referencing CMakeLists 
from [planning_scene_interface/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/planning_scene_interface/CMakeLists.txt) and [move_group_interface/CMakeLists.txt](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/move_group_interface/CMakeLists.txt)

I fix variables like below and error「CMake Error at py_bindings_tools/CMakeLists.txt:28 (install):
  install DIRECTORY given no DESTINATION!」disappeared.
```
CATKIN_GLOBAL_BIN_DESTINATION => bin
CATKIN_GLOBAL_INCLUDE_DESTINATION => include
catkin_LIBRARIES => なし
${CATKIN_DEVEL_PREFIX}/${CATKIN_PACKAGE_PYTHON_DESTINATION} => bin
```
Then I got a new error like this:
```
/usr/include/boost/python/detail/wrap_python.hpp:57:11: fatal error: pyconfig.h:No such file or directory. 
```
slove it by two steps:
1. use command 「find /usr/include -name pyconfig.h」to find where 「pyconfig.h」is.
   result（my system）:
  ```
/usr/include/x86_64-linux-gnu/python3.8/pyconfig.h
/usr/include/python3.8/pyconfig.h
```
2. set path in「CPLUS_INCLUDE_PATH」
  ```
export CPLUS_INCLUDE_PATH=/usr/include/python3.8/
(echo "export CPLUS_INCLUDE_PATH=/usr/include/python3.8/" >> ~/.bashrc)
  ```
Now I am fixing py_bindings_tools/src/roscpp_initializer.cpp 
Replacing the parts that are written with ros/ros.h to rclcpp/rclcpp.hpp

I have a little problem beacuse ros::AsyncSpinner() is used and never seen before.
but I find a solution in this two links：
[support for AsyncSpinner #335  ](https://github.com/ros2/rclcpp/issues/335)
[Concurrency and parallelism in ROS 1 and ROS 2](https://nicolovaligi.com/concurrency-and-parallelism-in-ros1-and-ros2-application-apis.html)
I will try it and report the result later.

chaos4ros2, 2021-04-27:
Progress report：

| no | origin | fix | reason | file | reference |
| --- | ---  | ---  |  ---  | --- | --- |
| 1   | ros::AsyncSpinner  | rclcpp::executors::MultiThreadedExecutor  |  ros1 => ros2 | [roscpp_initializer.cpp ](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/py_bindings_tools/src/roscpp_initializer.cpp#L142)| [link](https://github.com/ros2/examples/blob/master/rclcpp/executors/multithreaded_executor/multithreaded_executor.cpp#L162) |
| 2   | ros::serialization  | rclcpp::Serialization  |  ros1 => ros2  | [serialize_msg.h](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/py_bindings_tools/include/moveit/py_bindings_tools/serialize_msg.h#L111)  | [link](https://answers.ros.org/question/371866/ros1-vs-ros2-serialization/) |
| 3   | getSharedRobotModel(robot_description)  | getSharedRobotModel(node, robot_description)  |  [getSharedRobotModel()](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp#L116) has two arguments, not one  | [wrap_python_robot_interface.cpp](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/robot_interface/src/wrap_python_robot_interface.cpp#L75) | --- |
| 4   | bp::list  | boost::python::list  | namespace not work | [wrap_python_robot_interface.cpp](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/robot_interface/src/wrap_python_robot_interface.cpp#L58) | --- |
| 5   | geometry_msgs/PointStamped.h | geometry_msgs/msg/PointStamped.hpp  | ros1 => ros2 | [moveit_cpp_test.cpp](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/test/moveit_cpp_test.cpp#L54) | --- |
| 6   | visualization_msgs/MarkerArray.h| visualization_msgs/msg/MarkerArray.hpp | ros1 => ros2 | [wrap_python_robot_interface.cpp](https://github.com/chaos4ros2/moveit2/blob/main/moveit_ros/planning_interface/robot_interface/src/wrap_python_robot_interface.cpp#L49) | --- |

I create a list about critical errors that occured during compiling moveit_ros/planning_interface.

The correction of No.1~No.4 fix compile error but I'm not sure that the fix is ​​correct.No.5 & No.6 are still not solved. the origin may cause a fatal error: 「geometry_msgs/PointStamped.h:No such file or directory」so I change it to ROS2 style.
I also tried to add package name in Cmakelists and use absolute path but the error still occurs.

chaos4ros2, 2021-05-05:
Progress report 3：

Package moveit_ros/planning_interface's compile was successful but didn't create a shared library file(.so)
and the package's scale is huge so I could't find the problem.

However I found a good example introduced in [ros-planning/moveit#900](https://github.com/ros-planning/moveit/issues/900#issue-323907391) and finally create a [basic package](https://github.com/chaos4ros2/moveit2_python) based on [this commit](https://github.com/ros-planning/moveit_task_constructor/pull/99/commits/226b0b03b4a3b44fab588250352b6cc430769f59). this issue's goal is to run moveit2 [demos](https://github.com/ros-planning/moveit2/tree/main/moveit_commander/demos) that use python interface (pick.py or plan.py) so my next step is to pick up a easy one and make it executable.

closed 2022-06-02 by henningkayser