moveit/moveit2 #431: Standardize parameter parsing

AndyZe, 2021-04-20:
Particularly, `moveit_servo/src/servo_parameters.cpp` should be standardized.

I think it's a good idea to ignore int's getting cast as doubles, but throw a warning if the parameter type does not match expected otherwise.

henningkayser, 2021-05-04:
@AndyZe thanks for filing this. I was actually about do create a similar issue to finally get #313 going. For handling parameters, @tylerjw and I have already started [discussing some design ideas](https://github.com/PickNikRobotics/rosparam_shortcuts/pull/17) for a unified config API that could be applied to MoveIt or other ROS 2 packages.

aaroosh-07, 2021-07-02:
can i take up this issue

henningkayser, 2023-01-11:
This issue has become obsolete with the efforts being implemented with #335. The Servo refactoring in particular has its own issue that builds on top of param refactoring: #917. I'm closing this.

closed 2023-01-11 by henningkayser