moveit/moveit2 #432: Program stuck in moveit::planning_interface::MoveGroupInterface move_group

vtr101, 2021-04-20:
Hello,

I've converted the launch and config files from the setup assistant to a ROS2 format using the move group demo as an example and I used that package as a base to using my manipulator with moveit. However, the program hangs on moveit::planning_interface::MoveGroupInterface move_group(move_group_node, PLANNING_GROUP). The package is almost identical to the one used in the demo and my robot model loads in rviz; however, due to the program being stuck, as mentioned, I get errors such as: '[moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?' and can't plan a trajectory.

Does anyone also have this issue and/or can help me with it?

Thank you.

swilcock0, 2021-04-20:
Hey @vtr101 , is it possible to share a link to your launch file and c++? Is get_planning_scene listed if you run `ros2 service list` or is it possibly given a namespace prefix?

vtr101, 2021-04-21:
Hi, thanks for the response!

Here are my launch and c++ files: https://github.com/vtr101/blank

The get_planning_scene service is not listed at all.

swilcock0, 2021-04-21:
@vtr101 Have you got a customised version of the moveit repo? Line 96 of [arm4agrob_4Manipulator.launch.py](https://github.com/vtr101/blank/blob/main/arm4agrob_4Manipulator.launch.py) points to the `arm4agrob_moveit2` package rather than `moveit_ros_move_group`. This is likely messing with move group loading its capabilities - [Move Group uses the plugin loader](https://github.com/ros-planning/moveit2/blob/2be91d28b894b649e2012b23597a9a60590de1e9/moveit_ros/move_group/src/move_group.cpp#L124) with `moveit_ros_move_group`.

My personal advice would be to maintain the moveit naming and structure, source it and overlay with your own workspace

vtr101, 2021-04-21:
I was pointing to the wrong package the whole time. It's solved now! Thank you!

closed 2021-04-21 by vtr101