moveit/moveit2 #438: [ROS2 Foxy] fail to build from source E: Unable to locate package ros-foxy-moveit-core

newcanopies, 2021-04-25:
### Description

Installing moveit2 by [instructions](http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html) for Foxy  

$` rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`
 
`apt: Failed to detect successful installation of ros-foxy-moveit- .............`

See logs below.

### environment
* ROS2 Foxy
*  Ubuntu 20.04 (WSL2)
*  Binary
* [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main

### Actual behaviour

$` rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`
```
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-common] failed
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-core] failed
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-ros-planning] failed
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-ros-planning-interface] failed
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-ros-perception] failed
  apt: command [sudo -H apt-get install -y ros-foxy-moveit-fake-controller-manager] failed
  apt: Failed to detect successful installation of [ros-foxy-moveit-common]
  apt: Failed to detect successful installation of [ros-foxy-moveit-core]
  apt: Failed to detect successful installation of [ros-foxy-moveit-ros-planning]
  apt: Failed to detect successful installation of [ros-foxy-moveit-ros-planning-interface]
  apt: Failed to detect successful installation of [ros-foxy-moveit-ros-perception]
  apt: Failed to detect successful installation of [ros-foxy-moveit-fake-controller-manager]
```

### Backtrace or Console output

Troubleshooting:

$` grep -h ^deb /etc/apt/sources.list /etc/apt/sources.list.d/*`

```
deb http://archive.ubuntu.com/ubuntu/ focal main restricted
deb http://archive.ubuntu.com/ubuntu/ focal-updates main restricted
deb http://archive.ubuntu.com/ubuntu/ focal universe
deb http://archive.ubuntu.com/ubuntu/ focal-updates universe
deb http://archive.ubuntu.com/ubuntu/ focal multiverse
deb http://archive.ubuntu.com/ubuntu/ focal-updates multiverse
deb http://archive.ubuntu.com/ubuntu/ focal-backports main restricted universe multiverse
deb http://security.ubuntu.com/ubuntu/ focal-security main restricted
deb http://security.ubuntu.com/ubuntu/ focal-security universe
deb http://security.ubuntu.com/ubuntu/ focal-security multiverse
deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
deb http://packages.ros.org/ros/ubuntu focal main
deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main focal main
```

$ `apt-get update`

```
Reading package lists... Done
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
```

newcanopies, 2021-04-25:
https://gist.github.com/aliabidzaidi/fd873021efc7f70d276a7128ce22848e
did not resolve.

tylerjw, 2021-04-25:
> $ `apt-get update`

you need to always run this with sudo (unless you are the root user when you run it).

```bash
sudo apt-get update
```

tylerjw, 2021-04-25:
Also, what directory are you running the `rosdep install` command from?  The `--ignore-src` should mean that it won't try to install packages that it finds a source version of in the `--from-paths .` paths.  Assuming you cloned the moveit2 repos, those are all packages it should find source versions of.

newcanopies, 2021-04-25:
>  what directory are you running the `rosdep install` command from?

from colcon workspace  `cd ~/ws_moveit2/src`

Just removed the install, and tried again - same error at building colcon workspace with `rosdep install`

clone step:
_wget https://raw.githubusercontent.com/ros-planning/moveit2/main/moveit2.repos
vcs import < moveit2.repos
wget https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/moveit2_tutorials.repos
vcs import < moveit2_tutorials.repos_

does this look like a complete clone? @tylerjw (thank you!)

user@machine ~/ws_moveit2/src$ ll
```
total 60
drwxr-xr-x 12 esofi esofi 4096 Apr 25 23:25 ./
drwxr-xr-x  3 esofi esofi 4096 Apr 25 23:23 ../
drwxr-xr-x  8 root  root  4096 Apr 25 23:23 geometric_shapes/
-rw-r--r--  1 root  root   862 Apr 25 23:23 moveit2.repos
drwxr-xr-x  8 root  root  4096 Apr 25 23:27 moveit2_tutorials/
-rw-r--r--  1 esofi esofi  255 Apr 25 23:25 moveit2_tutorials.repos
-rw-r--r--  1 root  root   255 Apr 25 23:25 moveit2_tutorials.repos.1
drwxr-xr-x  8 root  root  4096 Apr 25 23:23 moveit_msgs/
drwxr-xr-x 10 root  root  4096 Apr 25 23:25 moveit_resources/
drwxr-xr-x 13 root  root  4096 Apr 25 23:23 ros2_control/
drwxr-xr-x 14 root  root  4096 Apr 25 23:23 ros2_controllers/
drwxr-xr-x 10 root  root  4096 Apr 25 23:25 rviz_visual_tools/
drwxr-xr-x  8 root  root  4096 Apr 25 23:23 srdfdom/
drwxr-xr-x  6 root  root  4096 Apr 25 23:23 warehouse_ros/
drwxr-xr-x  9 root  root  4096 Apr 25 23:23 warehouse_ros_mongo/
```

newcanopies, 2021-04-25:
WSL2 bug?

newcanopies, 2021-04-25:
could it be that some of the branches in `moveit2.repos` and their sub/modules, are out of sync with Foxy?

repositories:
  moveit_msgs:
    type: git
    url: https://github.com/ros-planning/moveit_msgs
    version: ros2
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: ros2
  geometric_shapes:
    type: git
    url: https://github.com/ros-planning/geometric_shapes
    version: ros2
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control
    version: master
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers
    version: master
  warehouse_ros:
    type: git
    url: https://github.com/ros-planning/warehouse_ros
    version: ros2
  warehouse_ros_mongo:
    type: git
    url: https://github.com/ros-planning/warehouse_ros_mongo
    version: ros2

newcanopies, 2021-04-25:
which repos are these missing ros deps sourced from:

 ros-foxy-moveit-common
 ros-foxy-moveit-core
 ros-foxy-moveit-ros-planning
 ros-foxy-moveit-ros-planning-interface
 ros-foxy-moveit-ros-perception
 ros-foxy-moveit-fake-controller-manager

tylerjw, 2021-04-25:
Two things.  I see now that those instructions are missing something.  You need to download the moveit2 repo itself into the source directory.

```bash
git clone https://github.com/ros-planning/moveit2.git
```

That's where those packages come from.  If you want to make a PR against the website to add that to the step with the vcs commands please tag me and we can fix that.

Secondly, does apt work on WSL?  Why are you doing this as root?  (I see the directories are owned by root).  While ROS2 works on windows, moveit2 is still not building on windows as far as I know.  There is an open PR from some people at Microsoft that is helping us with that.  I'm not sure those instructions will work at all in WSL.

tylerjw, 2021-04-25:
I honestly don't know enough about WSL to know if these instructions will work.  For example, does it use microsoft compilers (like visual studio) or are they ones that come from gnu or clang?  How does the user account work in wsl?  How does the package manager apt work?

I don't develop for windows and while it is "Tier 1" according to OSRF for foxy, it is more that microsoft is putting engineering effort towards that goal (including helping us get moviet2 working on windows).  For your own sanity, unless you are very with Windows development and are willing to go into uncharted places to fix these things I'd suggest you just use an actual Ubuntu 20.04 install for working with ROS.

tylerjw, 2021-04-25:
> could it be that some of the branches in `moveit2.repos` and their sub/modules, are out of sync with Foxy?

The issue is that we removed the actual moveit2 repo itself from that file and you'll need to use git to manually clone it into your source directory.  I'm sorry about that.  We made this change because this file is meant to represent the source dependencies of the repo and should not contain the repo itself.

newcanopies, 2021-04-25:
> moveit2 is still not building on windows as far as I know
perhaps when running Windows ROS

> does apt work on WSL
it does

> That's where those packages come from. If you want to make a PR against the website to add that to the step with the vcs commands please tag me and we can fix that.

thanks @tylerjw , you mean a PR branch with:

----

# Clone moveit2 source repository:
`cd ~/moveit2_ws/src && git clone https://github.com/ros-planning/moveit2.git`

----

sadly: 
```
Summary: 24 packages finished [6min 34s]
  1 package failed: moveit_core
  5 packages aborted: effort_controllers joint_state_controller joint_trajectory_controller position_controllers velocity_controllers
  8 packages had stderr output: controller_interface controller_manager diff_drive_controller forward_command_controller geometric_shapes joint_state_broadcaster joint_trajectory_controller moveit_core
  24 packages not processed
```

newcanopies, 2021-04-25:
>  There is an open PR from some people at Microsoft that is helping us with that. I'm not sure those instructions will work at all in WSL.

how can I find the PR?

tylerjw, 2021-04-25:
No suprise there.  here is the PR for windows support: https://github.com/ros-planning/moveit2/pull/286  It is out of date and last time I talked with the author he is working to update it.

Unless you have a ton of experience developing for windows I would generally avoid trying to use windows for ros development.  Ubuntu is what everyone who works on this repo (and much of the ros ecosystem) uses so it will be much easier for you to use if you are just getting started.  That being said, it is possible to get it working and microsoft would be happy to know that there are people trying to do it.  You will just have a much harder time getting help with anything ros if you are using windows.

tylerjw, 2021-04-25:
To get that PR to work you'll probably need the old versions of the other repos (from around the time he last updated it).

tylerjw, 2021-04-25:
Here is a fork that at one point worked on windows: https://github.com/lilustga/moveit2/tree/builds_on_windows

tylerjw, 2021-04-25:
> does this match the known fail on windows?

no idea... I don't know if the author of the PR I linked has a similar setup to you (using wsl) or if there are other things.  Like I said before, getting this to work on windows is not really supported because (outside of the author of that PR) no one here develops on windows.  I attempted to get it to work once and gave up.  Generally I feel like windows is very unfriendly to linux developers this and it would be much better if they just encouraged users to install a VM when they want to use something developed for linux instead of offering this wsl thing that is sort of half linux, half windows.

tylerjw, 2021-04-25:
I'm not trying to discourage you from learning ROS or using MoveIt2, I'm just warning you that I probably can't help you much and you will probably run into issues we have no way of reproducing if you are on Windows.

newcanopies, 2021-04-26:
> For example, does it use microsoft compilers (like visual studio) or are they ones that come from gnu or clang?

it can use either :)

thanks for solving the title issue

closed 2021-04-26 by newcanopies

tylerjw, 2021-04-26:
> thanks for solving the title issue

One other thing I wanted to point out is we have a Discord channel if you'd like to chat with us about anything: https://discord.gg/RrySut8

If you do submit an issue to the tutorial github page or a PR for adding the instruction for cloning the repo ping me and we can get that fixed for others.

gavanderhoorn, 2021-04-26:
Perhaps for @tylerjw: WSL "virtualised" a Linux kernel (not really, it was more a re-implementation of the Linux Kernel API/ABI), WSL 2 uses HyperV to run an actual VM (not a full one necessarily).

In the end, a full userland of the Linux distribution of your choice can then be run, as-if it was running natively on your machine (it basically is/was).

It's all Linux (Ubuntu typically), so no Windows specific fixes or hacks are needed.

Theoretically, anything which works under Linux should work in WSL/WSL2. In reality, there were problems with GUIs, GPU acceleration, access to hw (still a problem) and a few other restrictions.

What @newcanopies is trying to do should work fine, and the problems should not be related to using WSL or WSL2. *Those* problems come up at a later point.

newcanopies, 2021-04-26:
> Ubuntu running in WSL (or any other Linux) is Ubuntu, so at the very least the build instructions should-just-work.

@gavanderhoorn that is the hope. Simultaneous access to Windows software has many valid use-cases. 

while this WSL build fork https://github.com/lilustga/moveit2/tree/builds_on_windows
does process significantly more moveit2 dependencies than the master source, it also does ultimately fail to build. 

I am trying to understand the patching method(s) used by @lilustga on the above Windows build PR, to try and patch it myself, but colcon build seems to fail in relation to compiler handling of namespaces among other things.

gavanderhoorn, 2021-04-26:
> while this WSL build fork https://github.com/lilustga/moveit2/tree/builds_on_windows
> does process significantly more moveit2 dependencies than the master source build, it also does ultimately fail to build.

Afaik @lilustga is an MS employee. IIRC, the changes he made were to make MoveIt2 build *on Windows*, using MSCV/Visual Studio.

Building in a WSL2 session/terminal wouldn't need those, as you're not using Visual Studio there.

newcanopies, 2021-04-26:
> to make MoveIt2 build _on Windows_, using MSCV/Visual Studio.

Right! Thank you @gavanderhoorn initially I suspected that PR was for ROS on Windows, but somehow got it conflated.

Thanks for the Discord link @tylerjw I will troubleshoot why 
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` 
is failing on my WSL2 following the official [Foxy instructions](http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html), and may open a new Issue if needed.

tylerjw, 2021-04-26:
I'm sorry if I misunderstood, I hope doing this is as straightforward as it is on ubuntu.

lilustga, 2021-04-26:
@newcanopies gavanderhoorn is correct that the changes are to build on Windows and not on WSL which would be building on whichever Linux kernel you are using. We have not tested moveit2 on WSL.

Following these instructions and installing ros-foxy-desktop will allow you to use Moveit2 on Windows:
https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html