moveit/moveit2 #440: colcon build problem at moveit_core

AtariaLee, 2021-04-26:
### Description

After  i update my moveit2 files today,some problems happend in build.When i run `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` to build my workspace,some problems happend.
 I dont know what happend,and how to solve it.I will be very grateful to you for your help!!!
### Your environment
* ROS Distro: RO2 FOXY
* OS Version: Ubuntu 20.04
* maybe binary?

### Steps to reproduce
When i run `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` to build my workspace,some problems happend.
### Expected behaviour
 All files are successful!

### Backtrace or Console output
--- stderr: moveit_core
CMake Error at utils/CMakeLists.txt:16 (add_library):
  Target "moveit_test_utils" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_robot_model" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  robot_model/CMakeLists.txt:35 (ament_add_gtest)

CMake Error at robot_model/CMakeLists.txt:4 (add_library):
  Target "moveit_robot_model" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_all_valid" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:48 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_world_diff" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:44 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_world" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:40 (ament_add_gtest)

CMake Error at collision_detection/CMakeLists.txt:3 (add_library):
  Target "moveit_collision_detection" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at kinematics_base/CMakeLists.txt:4 (add_library):
  Target "moveit_kinematics_base" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_transforms" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  transforms/CMakeLists.txt:20 (ament_add_gtest)

CMake Error at transforms/CMakeLists.txt:3 (add_library):
  Target "moveit_transforms" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cartesian_interpolator" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  robot_state/CMakeLists.txt:77 (ament_add_gtest)

And a lots of problems just like it.

tylerjw, 2021-04-26:
did you forget to source the foxy install?

newcanopies, 2021-04-26:
I believe I encountered the same issue attempting a first time build too @AtariaLee @tylerjw 

https://github.com/ros-planning/moveit/issues/2621
Full Gist https://gist.github.com/newcanopies/c1c95cc27ec4f8dd7a002572e6849d8c

tylerjw, 2021-04-26:
This is fixed by https://github.com/ros-planning/moveit2/pull/442

AtariaLee, 2021-04-27:
@tylerjw   Its really helpful ,thanks for your help!

closed 2021-04-27 by AtariaLee