moveit/moveit2 #441: colcon build fails at moveit_core can't find target (ROS2 Foxy)

newcanopies, 2021-04-26:
### Description

colcon fails to build moveit_core
find_package() calls seem to work but targets aren't picked up

sourced foxy install before build
$ . /opt/ros/foxy/setup.bash
$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release  

full output log is in [this Gist](https://gist.github.com/newcanopies/c1c95cc27ec4f8dd7a002572e6849d8c) and also below. 

### Your environment
* ROS2 Foxy
* Ubuntu 20.04 / WSL2
* Binary
* ?

### Backtrace or Console output

```
- \\ -

--- stderr: moveit_core
CMake Error at utils/CMakeLists.txt:16 (add_library):
  Target "moveit_test_utils" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_robot_model" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  robot_model/CMakeLists.txt:35 (ament_add_gtest)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< moveit_core [59.4s, exited with code 1]
Aborted  <<< diff_drive_controller [1min 27s]
Aborted  <<< joint_state_broadcaster [1min 33s]
Aborted  <<< position_controllers [20.5s]
Aborted  <<< velocity_controllers [20.6s]
Aborted  <<< joint_trajectory_controller [1min 39s]
Aborted  <<< effort_controllers [25.7s]

Summary: 22 packages finished [6min 9s]
  1 package failed: moveit_core
  6 packages aborted: diff_drive_controller effort_controllers joint_state_broadcaster joint_trajectory_controller position_controllers velocity_controllers
  8 packages had stderr output: controller_interface controller_manager diff_drive_controller forward_command_controller geometric_shapes joint_state_broadcaster joint_trajectory_controller moveit_core
  25 packages not processed
```

Full colcon build log is in this Gist: https://gist.github.com/newcanopies/c1c95cc27ec4f8dd7a002572e6849d8c

tylerjw, 2021-04-26:
I'm trying to reproduce this locally... I'll let you know what I find.

endre90, 2021-04-27:
Hi, colcon fails to build moveit_core for me as well. Here is the trace:

Starting >>> moveit_core
--- stderr: moveit_core                            
moveit_core: You did not request a specific build type: Choosing 'Release' for maximum performance
CMake Error at utils/CMakeLists.txt:16 (add_library):
  Target "moveit_test_utils" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_robot_model" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  robot_model/CMakeLists.txt:35 (ament_add_gtest)

CMake Error at robot_model/CMakeLists.txt:4 (add_library):
  Target "moveit_robot_model" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_all_valid" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:48 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_world_diff" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:44 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_world" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  collision_detection/CMakeLists.txt:40 (ament_add_gtest)

CMake Error at collision_detection/CMakeLists.txt:3 (add_library):
  Target "moveit_collision_detection" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at kinematics_base/CMakeLists.txt:4 (add_library):
  Target "moveit_kinematics_base" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
[... 374 lines cut ...]
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at kinematics_metrics/CMakeLists.txt:3 (add_library):
  Target "moveit_kinematics_metrics" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at dynamics_solver/CMakeLists.txt:3 (add_library):
  Target "moveit_dynamics_solver" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at utils/CMakeLists.txt:16 (add_library):
  Target "moveit_test_utils" links to target
  "resource_retriever::resource_retriever" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< moveit_core [4.55s, exited with code 1]

endre90, 2021-04-27:
Included the #442 fix, now some bugs are coming up. I pulled a release, still the same:

Starting >>> moveit_core
[Processing: moveit_core]                               
[Processing: moveit_core]                                        
[Processing: moveit_core]                                         
[Processing: moveit_core]                                         
--- stderr: moveit_core
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In function ‘bool collision_detection::distanceCallback(fcl::CollisionObjectd*, fcl::CollisionObjectd*, void*, double&)’:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:40: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                        ^~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:14: error: parse error in template argument list
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: no matching function for call to ‘static_pointer_cast<<expression error> >(const std::shared_ptr<const fcl::CollisionGeometry<double> >&)’
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
In file included from /usr/include/c++/9/bits/shared_ptr.h:52,
                 from /usr/include/c++/9/memory:81,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/declare_ptr.h:37,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/world.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:37:
/usr/include/c++/9/bits/shared_ptr_base.h:1532:5: note: candidate: ‘template<class _Tp, class _Tp1, __gnu_cxx::_Lock_policy _Lp> std::__shared_ptr<_Tp1, _Lp> std::static_pointer_cast(const std::__shared_ptr<_Tp2, _Lp>&)’
 1532 |     static_pointer_cast(const __shared_ptr<_Tp1, _Lp>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr_base.h:1532:5: note:   template argument deduction/substitution failed:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: template argument 1 is invalid
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/declare_ptr.h:37,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/world.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:37:
/usr/include/c++/9/bits/shared_ptr.h:491:5: note: candidate: ‘template<class _Tp, class _Up> std::shared_ptr<_Tp> std::static_pointer_cast(const std::shared_ptr<_Tp>&)’
  491 |     static_pointer_cast(const shared_ptr<_Up>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:491:5: note:   template argument deduction/substitution failed:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: template argument 1 is invalid
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:40: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
  525 |        !std::static_pointer_cast<const fcl::OcTreed>(o2->collisionGeometry())->getRoot()))
      |                                        ^~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:14: error: parse error in template argument list
  525 |        !std::static_pointer_cast<const fcl::OcTreed>(o2->collisionGeometry())->getRoot()))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:77: error: no matching function for call to ‘static_pointer_cast<<expression error> >(const std::shared_ptr<const fcl::CollisionGeometry<double> >&)’
[... cut ...]
make[2]: *** [collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/build.make:63: collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/src/collision_common.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1615: collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [2min 1s, exited with code 2]

Summary: 68 packages finished [2min 28s]
  1 package failed: moveit_core

newcanopies, 2021-04-27:
> Included the #442 fix, now some bugs are coming up. I pulled a release, still the same:

hi @endre90 the #442 addressed some up-stream changes in dependencies, and built successfully yesterday.

make sure you pull #422 from here: $ `cd /home/endre/foxy_ws/src/moveit2/moveit_core/ `
since **moveit_core** is failing.

endre90, 2021-04-27:
Hi, thanks for the details tag, neat trick :).
Unfortunately, I have no success building moveit_core.

As I wrote, I updated moveit_core according to #442, and got build errors as listed in https://github.com/ros-planning/moveit2/issues/441#issuecomment-827596058.

Should I open a new issue?

Best Regards,
Endre

newcanopies, 2021-04-27:
Np! @endre90 

Upto you. 
Based on the discussion in PR #442 , it is likely related to `ament` / `moveit_core` work in progress on that PR. 
Perhaps comment with a `Gist` of your Log in there first and @ the developers? 
Or make new Issue for https://github.com/ros-planning/moveit2/issues/441#issuecomment-827596058

endre90, 2021-04-27:
Yes, thanks, that's likely the case. New issue it is.

tylerjw, 2021-04-27:
I closed the attached PR as I fixed it at the source in geometric_shapes.  Please pull the latest from that repo for the fix.

closed 2021-04-27 by tylerjw