moveit/moveit2 #443: /moveit_core/collision_detection_fcl build fails

endre90, 2021-04-27:
### Description
moveit_core fails building at package collision_detection_fcl.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source: main branch

### Steps to reproduce
clone moveit2
fix issue #442 in moveit_core
colcon build

### Backtrace or Console output
<details>
```
--- stderr: moveit_core                                          
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In function ‘bool collision_detection::distanceCallback(fcl::CollisionObjectd*, fcl::CollisionObjectd*, void*, double&)’:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:40: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                        ^~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:14: error: parse error in template argument list
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: no matching function for call to ‘static_pointer_cast<<expression error> >(const std::shared_ptr<const fcl::CollisionGeometry<double> >&)’
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
In file included from /usr/include/c++/9/bits/shared_ptr.h:52,
                 from /usr/include/c++/9/memory:81,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/declare_ptr.h:37,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/world.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:37:
/usr/include/c++/9/bits/shared_ptr_base.h:1532:5: note: candidate: ‘template<class _Tp, class _Tp1, __gnu_cxx::_Lock_policy _Lp> std::__shared_ptr<_Tp1, _Lp> std::static_pointer_cast(const std::__shared_ptr<_Tp2, _Lp>&)’
 1532 |     static_pointer_cast(const __shared_ptr<_Tp1, _Lp>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr_base.h:1532:5: note:   template argument deduction/substitution failed:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: template argument 1 is invalid
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
In file included from /usr/include/c++/9/memory:81,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/declare_ptr.h:37,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/world.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:37:
/usr/include/c++/9/bits/shared_ptr.h:491:5: note: candidate: ‘template<class _Tp, class _Up> std::shared_ptr<_Tp> std::static_pointer_cast(const std::shared_ptr<_Tp>&)’
  491 |     static_pointer_cast(const shared_ptr<_Up>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:491:5: note:   template argument deduction/substitution failed:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:523:77: error: template argument 1 is invalid
  523 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) ||
      |                                                                             ^
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:40: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
  525 |        !std::static_pointer_cast<const fcl::OcTreed>(o2->collisionGeometry())->getRoot()))
      |                                        ^~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:14: error: parse error in template argument list
  525 |        !std::static_pointer_cast<const fcl::OcTreed>(o2->collisionGeometry())->getRoot()))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:525:77: error: no matching function for call to ‘static_pointer_cast<<expression error> >(const std::shared_ptr<const fcl::CollisionGeometry<double> >&)’
  525 |        !std::static_pointer_cast<const fcl::OcTreed>(o2->collisionGeometry())->getRoot()))
      |                                                                             ^
In file included from /usr/include/c++/9/bits/shared_ptr.h:52,
                 from /usr/include/c++/9/memory:81,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/declare_ptr.h:37,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/world.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:39,
                 from /home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:37:
[... 23 lines cut ...]
      |                  ^~~
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:842:17: error: expected ‘;’
  842 |       cg_g = new fcl::OcTreed(g->octree);
      |                 ^~~~
      |                 ;
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In instantiation of ‘collision_detection::FCLGeometryConstPtr collision_detection::createCollisionGeometry(const ShapeConstPtr&, const T*, int) [with BV = fcl::OBBRSS<double>; T = moveit::core::LinkModel; collision_detection::FCLGeometryConstPtr = std::shared_ptr<const collision_detection::FCLGeometry>; shapes::ShapeConstPtr = std::shared_ptr<const shapes::Shape>]’:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:864:97:   required from here
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:841:29: warning: unused variable ‘g’ [-Wunused-variable]
  841 |       const shapes::OcTree* g = static_cast<const shapes::OcTree*>(shape.get());
      |                             ^
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In instantiation of ‘collision_detection::FCLGeometryConstPtr collision_detection::createCollisionGeometry(const ShapeConstPtr&, const T*, int) [with BV = fcl::OBBRSS<double>; T = moveit::core::AttachedBody; collision_detection::FCLGeometryConstPtr = std::shared_ptr<const collision_detection::FCLGeometry>; shapes::ShapeConstPtr = std::shared_ptr<const shapes::Shape>]’:
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:870:98:   required from here
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:841:29: warning: unused variable ‘g’ [-Wunused-variable]
/home/endre/foxy_ws/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In instantiation of ‘collision_detection::FCLGeometryConstPtr collision_detection::createCollisionGeometry(const ShapeConstPtr&, const T*, int) [with BV = fcl::OBBRSS<doub [... cut ...]
make[2]: *** [collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/build.make:63: collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/src/collision_common.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1615: collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [2min 6s, exited with code 2]
```
<\details>

tylerjw, 2021-04-27:
What compiler are you using?  I only asking because I'm not seeing those errors locally.

tylerjw, 2021-04-27:
I'm on 20.04 and this is the compiler I'm using:

```
gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

endre90, 2021-04-27:
Hi, same for me:
```$ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.```

tylerjw, 2021-04-27:
These errors all seem to be related to fcl.  The first error seems to be about `const fcl::OcTreed` for example.  What version of libfcl-dev do you have installed:

```
apt info libfcl-dev
Package: libfcl-dev
Version: 0.5.0-5build1
Priority: extra
Section: universe/libdevel
Source: fcl
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 889 kB
Depends: pkg-config, libboost-filesystem-dev, libboost-test-dev, libccd-dev, libtinyxml-dev, libflann-dev, liboctomap-dev, libfcl0.5 (= 0.5.0-5build1)
Homepage: http://gamma.cs.unc.edu/FCL/
Download-Size: 88.4 kB
APT-Manual-Installed: yes
APT-Sources: http://us.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
Description: Flexible Collision Library - development files
 FCL is a collision and proximity library that integrates several techniques for
 fast and accurate collision checking and proximity computation. Tje library is
 based on hierarchical representations and designed to perform multiple
 proximity queries on different model representations. The set of queries
 includes discrete collision detection, continuous collision detection,
 separation distance computation and penetration depth estimation. The input
 models may correspond to triangulated rigid or deformable models and
 articulated models. This package contains the development files.
```

endre90, 2021-04-27:
Hi, same for me:

```
Package: libfcl-dev
Version: 0.5.0-5build1
Priority: extra
Section: universe/libdevel
Source: fcl
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 889 kB
Depends: pkg-config, libboost-filesystem-dev, libboost-test-dev, libccd-dev, libtinyxml-dev, libflann-dev, liboctomap-dev, libfcl0.5 (= 0.5.0-5build1)
Homepage: http://gamma.cs.unc.edu/FCL/
Download-Size: 88.4 kB
APT-Manual-Installed: no
APT-Sources: http://se.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
Description: Flexible Collision Library - development files
 FCL is a collision and proximity library that integrates several techniques for
 fast and accurate collision checking and proximity computation. Tje library is
 based on hierarchical representations and designed to perform multiple
 proximity queries on different model representations. The set of queries
 includes discrete collision detection, continuous collision detection,
 separation distance computation and penetration depth estimation. The input
 models may correspond to triangulated rigid or deformable models and
 articulated models. This package contains the development files.
```

tylerjw, 2021-04-27:
If you run this in your workspace what is the output:
```grep -r "CMAKE_CXX_COMPILER_ID" build```

I get a bunch of copies of this:
```set(CMAKE_CXX_COMPILER_ID "GNU")```

I'm trying to figure out if you are using gcc or something else.

I just updated all the packages on my system, did a rosdep install and and a clean build and still can't reproduce this.

tylerjw, 2021-04-27:
Are you just running `colcon build` from the workspace directory or are you giving any arguments?

I typically use mixins and am building like this:  `colcon build --mixin release ccache` but if I build without any arguments I still don't see any errors (other than the warnings that geometric_shapes) produces.

endre90, 2021-04-27:
This is what I get:

https://gist.github.com/endre90/563000312e66aef49dea4d7def76b07f

tylerjw, 2021-04-27:
ok, you are using the same compiler as me.  what is the colcon command you are building with?

endre90, 2021-04-27:
colcon build

vatanaksoytezer, 2021-04-27:
@endre90 May be unrelated but can you try building with Release configuration? For example you can use the following command to build. Also sometimes (at least for me) some errors stuck and I usually clean my ws with `rm -rf build/ install / log/`

`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

endre90, 2021-04-28:
Hi, thanks for the tip, however I already tried it.
Still, collision_detection_fcl won't build due to https://gist.github.com/endre90/0f231fa2693190ca739227cf04b280b2.
Quite strange if I am the only person having this issue, which means that there should be something wrong at my end. 
What I have done is updated and upgraded everything a few days ago, could it somehow be connected to this?
https://github.com/ros2/ros2/issues/1074

endre90, 2021-04-28:
Hi again and thanks for the tip.

I already did that 2 times, but just for good measure, I did it again and it resorts in https://gist.github.com/endre90/0f231fa2693190ca739227cf04b280b2. It really bugs me how can this be if we have the same setup and are doing the same things. I think at this point I will just try to "fix" the C++ code myself, however it doesn't look to me like it needs fixing. I don't know why the compiler fails building with these errors.

Best,
Endre

newcanopies, 2021-04-28:
Maybe some clues can be found in the colcon build generated `moveit2/log/latest_build/logger_all.log`
Good luck @endre90

endre90, 2021-04-28:
Fixed. Compatibility issue with some C++ shared libraries. I didn't really manage to solve it in a nice way since every change produced other compatibility issues, so the sledgehammer approach worked: fresh Ubuntu installation.

Anyway, thanks for all the comments.

Best,
Endre

closed 2021-04-28 by endre90

tylerjw, 2021-04-28:
> sledgehammer approach worked: fresh Ubuntu installation.

ouch, I'm sorry.  That's never a nice solution.  That explains why the rest of us couldn't reproduce it though.

Volksfest, 2023-10-24:
Well, maybe two years too late but I also got the same problem; so you weren't alone.
The problem was probably building FCL on your own (at least it is what I did) without having octomap built first. There is no error only "optional" omitted build of octomaps in FCL which are not so optional in moveit...
Honestly, I still have no idea why FCL is not a ROS package but has a "optional" dependency to a ROS package (octomap)...

raghavvs, 2023-11-08:
Hi @Volksfest, do you mind sharing how you fixed it? I already have `octomap` built but I still have issues when building `/moveit_core/collision_detection_fcl`. 

````
--- stderr: moveit_core                                 
/home/raghav/ws_moveit2/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp: In function ‘bool collision_detection::distanceCallback(fcl::CollisionObjectd*, fcl::CollisionObjectd*, void*, double&)’:
/home/raghav/ws_moveit2/src/moveit2/moveit_core/collision_detection_fcl/src/collision_common.cpp:564:40: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
  564 |        !std::static_pointer_cast<const fcl::OcTreed>(o1->collisionGeometry())->getRoot()) || 
```

Thanks!

Volksfest, 2023-11-11:
@raghavvs Did you built FCL yourself? Did you built it such that cmake finds your octomap installation? I assume your FCL is NOT built against octomap and as such it does not define fcl::OcTreed.