moveit/moveit2 #444: Motion planning cannot avoid Octomap in planning scene

LossOfMe, 2021-04-28:
### Description
I am working on robot motion planning with perception from point cloud topic (3d camera). I found that when I planned robot path, the robot cannot avoid the collision in planning scene as shown in Figure [image].

I tried to solve this problem and found that one of the solution is it need to add something to planning scene using Scene objects tab UI and push Publish button as shown in Figure [image]

After that I planned robot path again and you can see robot can avoid the obstacle 
[image].

Do anyone have the way to solve this problem or the way to check this issue?

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04

swilcock0, 2021-04-29:
This looks like #407 and #429

henningkayser, 2021-05-06:
This doesn't look like #407 to me, but it might really be related to #429 which also looks like planning scene updates not being synced properly. I'll check if there is something odd going on, collision checking working after the manual update is very suspicious.

swilcock0, 2021-05-06:
@henningkayser the manual update thing is the same with #407

PrakarnJ, 2021-05-10:
@JafarAbdi Yeah It can fixed.

closed 2021-05-21 by PrakarnJ