moveit/moveit2 #446: Build issues on some packages on macOS

nkalupahana, 2021-05-01:
### Description

Most of my MoveIt2 build from source works fine. For instance:
```
bash-3.2$ colcon build --packages-select moveit_core
Starting >>> moveit_core
Finished <<< moveit_core [13.4s]                     

Summary: 1 package finished [13.7s]
```

However, there are certain packages that have serious issues when compiling; for instance:

```
bash-3.2$ colcon build --packages-select moveit_ros_warehouse
Starting >>> moveit_ros_warehouse
--- stderr: moveit_ros_warehouse                         
moveit_ros_warehouse: You did not request a specific build type: Choosing 'Release' for maximum performance
CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
[... 206 lines cut ...]
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< moveit_ros_occupancy_map_monitor [4.51s, exited with code 2]
```

What is going on here? I think it might have something to do with Boost, but I'm really not sure.

### Your environment
* ROS Distro: Foxy
* OS Version: macOS 11
* Source build (main)

swilcock0, 2021-05-03:
Hey @nkalupahana, how about if you try installing the packages with (for example) `colcon build --packages-up-to moveit_ros_planning_interface` to ensure the dependencies are built? Does the recommended 
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`
work for building the full package?

nkalupahana, 2021-05-03:
@swilcock0 First command fails with same error message:
```
bash-3.2$ colcon build --packages-up-to moveit_ros_planning_interface
Starting >>> eigen_stl_containers
Starting >>> object_recognition_msgs
Starting >>> octomap_msgs
Starting >>> random_numbers
Starting >>> angles
Starting >>> moveit_common                                          
Starting >>> srdfdom
Starting >>> moveit_resources_fanuc_description
Finished <<< eigen_stl_containers [2.93s]                                   
Starting >>> moveit_resources_panda_description                              
Finished <<< moveit_common [2.92s]                                           
Starting >>> control_msgs
Finished <<< angles [3.06s]                                                  
Finished <<< moveit_resources_fanuc_description [3.03s]
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> moveit_resources_pr2_description                                
Finished <<< random_numbers [3.45s]                                          
Starting >>> geometric_shapes
--- stderr: srdfdom                                                          
CMake Deprecation Warning at /Users/nisala/ros2-osx/src/gtest_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

---
Finished <<< srdfdom [4.76s]
Starting >>> warehouse_ros
Finished <<< moveit_resources_panda_description [2.97s]                      
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_moveit_config [2.97s]                    
Finished <<< moveit_resources_pr2_description [2.99s]                        
Finished <<< moveit_resources_panda_moveit_config [2.99s]                    
Finished <<< warehouse_ros [7.13s]      
--- stderr: geometric_shapes            
geometric_shapes: You did not request a specific build type: Choosing 'Release' for maximum performance
CMake Deprecation Warning at /Users/nisala/ros2-osx/src/gtest_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

---
Finished <<< geometric_shapes [9.09s]
Finished <<< octomap_msgs [16.5s]       
Finished <<< object_recognition_msgs [21.0s]
Starting >>> moveit_msgs
Finished <<< control_msgs [25.4s]                                      
[Processing: moveit_msgs]                                
[Processing: moveit_msgs]                                      
[Processing: moveit_msgs]                                       
Finished <<< moveit_msgs [1min 40s]                             
Starting >>> moveit_core
Finished <<< moveit_core [23.3s]                              
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
--- stderr: moveit_ros_occupancy_map_monitor 
moveit_ros_occupancy_map_monitor: You did not request a specific build type: Choosing 'Release' for maximum performance
CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
[... 32 lines cut ...]
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /Users/nisala/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< moveit_ros_occupancy_map_monitor [6.79s, exited with code 2]
```

nkalupahana, 2021-05-03:
Okay, got it fixed. It looks like when I was running into some other package issues, I ran build with `--cmake-force-configure`, which seems to have caused the CMake issues. Deleting and restarting the whole installation process fixed the issue.

closed 2021-05-03 by nkalupahana