moveit/moveit2 #453: Start state is ignored (default values are used) when planner is invoked

prajd3d, 2021-05-06:
### Description

Calls to the planner ignore joint values of other move groups in the start state of the robot. Planner assumes the default values for other move groups. In my case, the default values are in self-collision, so planning always fails.
 
After my debugging session, I have traced the problem down to one single line of code in  "threadsafe_state_storage.cpp":

`st->setToDefaultValues();`

It seems to me that this line should not be there as it negates intention of intializing the robot state the line before:

`st = new moveit::core::RobotState(start_state_)`

Please see the steps to reproduce the issue.

### Your environment
* ROS Distro: ROS2 Foxy
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? foxy

### Steps to reproduce

I use a robot (R) with multiple move groups (e.g., MG1, MG2).
Due to the nature of how the default configuration is computed by the RobotState class, the default configuration is in self-collision.

I initialize the R to a state q0 which sets joint values of MG1 and MG2. q0 is known to be collision-free.

Then, I initiate a plan request for just moving a single joint in MG1 by a very small value. Let the new target state be q1. q1 is also known to be collision-free.

The path from q0 to q1 is trivial and guaranteed to be collision free.
```

robot_model_loader::RobotModelLoader robot_model_loader(node_ptr, "robot_description");
auto model = robot_model_loader.getModel();
// For the failure to happen, the default configuration of the robot should be in collision.

auto scene = std::make_shared<planning_scene::PlanningScene>(model);
auto start_state = scene->getCurrentState();

moveit_msgs::msg::MotionPlanRequest req;
req.group_name = "MG1";
req.planner_id = "";
req.num_planning_attempts = 1;
req.allowed_planning_time = 5.0; 
req.max_velocity_scaling_factor =1;
req.max_acceleration_scaling_factor = 1;
moveit::core::robotStateToRobotStateMsg(
    start_state, req.start_state, true);

// Create goal_state by just changing a single joint in the move group
auto joint_model_group = model->getJointModelGroup("MG1");
goal_state = ... ;
moveit_msgs::msg::Constraints constraints = kinematic_constraints::constructGoalConstraints(goal_state, joint_model_group);

req.goal_constraints.push_back(constraints);

moveit_msgs::msg::MotionPlanResponse res;
auto pipeline = planning_pipelines_.at(plan_request_parameters.planning_pipeline);
pipeline->generatePlan(scene, req, res);
```

### Expected behaviour
Plan request should succeed.

### Actual behaviour
The plan request fails with the message:

"Skipping invalid start state (invalid state)"

### Backtrace or Console output

[ERROR] [1620258312.425252036] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - XXXXXXXX: Motion planning start tree could not be initialized!
[INFO] [1620258312.425263149] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:140 - No solution found after 0.000430 seconds
[WARN] [1620258312.433765028] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/goals/src/GoalLazySamples.cpp:129 - Goal sampling thread never did any work.
[INFO] [1620258312.433860892] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem

JafarAbdi, 2021-05-06:
> auto scene = std::make_shared<planning_scene::PlanningScene>(model);
auto start_state = scene->getCurrentState();

You are constructing a planning scene object which will have the default joint values for the internal robot state (it will not correspond to the state in RViz), you need to use the planning scene monitor to sync the state

prajd3d, 2021-05-06:
Thank you for your time. You are right in your comment. However, I did a bad job of explaining the problem. The code snippet that I attached earlier is unclear and does not explain the problem well. The problem is not the "getCurrentState()".

In my code (which I have on my codebase), I appropriately set the scene's current state before making any queries (code snippet is updated now)

I have updated the code snippet to reflect that. Could you please reopen the issue? The bug that is happening with TSStorage (threadsafe_state_storage.cpp) is subtle. The issue only shows up when the following is true:
1. Robot has a default state that is in collision
2. Multiple move groups are setup
3. A valid planning query targets one of the move-groups ( valid => path exists between q0 and q1 and RRTConnect can find it before timeout)

JafarAbdi, 2021-05-06:
@prajd3d Could you please share the whole console output .?

prajd3d, 2021-05-06:
> @prajd3d Could you please share the whole console output .?

I have many robots, so I had to snip the output.

Parsing robot urdf xml string.
[INFO] [1620326288.937600616] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00572771 seconds
[INFO] [1620326288.937619355] [moveit_robot_model.robot_model]: Loading robot model 'assembly_cell'...
... <snipped here>
Link R17_link_3 had 1 children
Link R17_link_4 had 1 children
Link R17_link_5 had 1 children
Link R17_link_6 had 1 children
Link R17_tool0 had 1 children
...<snipped here>

[INFO] [1620326288.966417812] [motion_planning_commons.planner]: Loading planning pipeline 'ompl'
[INFO] [1620326288.969182169] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[INFO] [1620326288.971458726] [moveit_ros.fix_workspace_bounds]: Param 'default_workspace_bounds' was not set. Using default value: 10.000000
[INFO] [1620326288.971485943] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[INFO] [1620326288.971500158] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[INFO] [1620326288.971505400] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[INFO] [1620326288.971509795] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[INFO] [1620326288.971513794] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[INFO] [1620326288.988568919] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'R11_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
**[WARN] [1620326288.989099251] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - R11_arm/R11_arm: Skipping invalid start state (invalid state)**
**[ERROR] [1620326288.989136536] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - R11_arm/R11_arm: Motion planning start tree could not be initialized!**
[WARN] [1620326288.989215559] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - R11_arm/R11_arm: Skipping invalid start state (invalid state)
[WARN] [1620326288.989242422] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - R11_arm/R11_arm: Skipping invalid start state (invalid state)
[ERROR] [1620326288.989270022] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - R11_arm/R11_arm: Motion planning start tree could not be initialized!
[ERROR] [1620326288.989354377] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - R11_arm/R11_arm: Motion planning start tree could not be initialized!
[WARN] [1620326288.989453504] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/src/Planner.cpp:257 - R11_arm/R11_arm: Skipping invalid start state (invalid state)
[ERROR] [1620326288.989481031] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:207 - R11_arm/R11_arm: Motion planning start tree could not be initialized!
[WARN] [1620326288.989665482] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 0.000855 seconds
[WARN] [1620326288.998627872] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/base/goals/src/GoalLazySamples.cpp:129 - Goal sampling thread never did any work.
[INFO] [1620326288.998716388] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[ERROR] [1620326288.998740827] [motion_planning_commons.planner]: Could not compute plan successfully.
/home/ros/workspace/ws_mp/src/motion_planning_commons/test/test_planner.cpp:76: Failure
Expected equality of these values:
  result->error_code.val
    Which is: 99999
  moveit_msgs::msg::MoveItErrorCodes::SUCCESS
    Which is: 1
[  FAILED  ] test_planner_basics.planning_1 (69 ms)
[----------] 2 tests from test_planner_basics (138 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (138 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] test_planner_basics.planning_1

 1 FAILED TEST

henningkayser, 2021-05-21:
Fixed with #455

closed 2021-05-21 by henningkayser