moveit/moveit2 #459: build error, transformToEigen

lorepieri8, 2021-05-07:
### Description

Cannot build successfully.

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Source  build
* 
### Steps to reproduce
Follow build instructions (https://moveit.ros.org/install-moveit2/source/) and type:
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Backtrace or Console output
[https://gist.github.com/lorepieri8/9266b13464e85c2ae5d3625aa5154301](url)

peterdavidfagan, 2021-05-18:
@lorepieri8 similar to the following issue (https://github.com/ros-planning/moveit2/issues/458),

I recently was able to build MoveIt2 from source using the instructions (http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html). My environment setup matches the description you provided above.

How did you install ROS2? Did you build ROS2 from source?

Note I used desktop installation of ROS2.

henningkayser, 2021-05-19:
Hey @lorepieri8, looking into your issue I couldn't find anything that might have caused it. I compile fresh workspaces with the same setup pretty frequently so there must be something specific to your setup that breaks this. Could you post a `vcs status` from your `src` directory? And you might want to try #465 to see if it helps.

lorepieri8, 2021-05-19:
I also have ros1 Noetic installed, not sure if that's causing issues. I've installed ros2 with `sudo apt install ros-foxy-desktop
`.

`~/ws_moveit2/src$ vcs status
............
=== ./fake_joint (git) ===
On branch foxy
Your branch is up to date with 'origin/foxy'.

nothing to commit, working tree clean
=== ./geometric_shapes (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
=== ./moveit2 (git) ===
HEAD detached at d95e4aecf
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	install/
	log/

nothing added to commit but untracked files present (use "git add" to track)
=== ./moveit_msgs (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
=== ./moveit_resources (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
=== ./ros2_control (git) ===
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
=== ./ros2_controllers (git) ===
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
=== ./srdfdom (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
=== ./urdfdom (git) ===
On branch foxy
Your branch is up to date with 'origin/foxy'.

nothing to commit, working tree clean
=== ./urdfdom_headers (git) ===
On branch foxy
Your branch is up to date with 'origin/foxy'.

nothing to commit, working tree clean
=== ./warehouse_ros (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
=== ./warehouse_ros_mongo (git) ===
On branch ros2
Your branch is up to date with 'origin/ros2'.

nothing to commit, working tree clean
`

henningkayser, 2021-05-21:
Ok, I just checked your attached log again and there seems to be a mix-up between ROS1 and ROS2 dependencies (see 'geometry_msgs::TransformStamped', that one should really not be in there). Also, the moveit2 repo includes build artifacts (install, logs directories) which might cause problems of they were sourced at some point. I would suggest clearing those up, starting a fresh shell that only sources foxy, wipe the ROS2 workspace and build from scratch.

JafarAbdi, 2021-06-30:
I'm going to close this, feel free to reopen it if you still have a problem building moveit2

closed 2021-06-30 by JafarAbdi

lorepieri8, 2021-08-14:
For future adventurers: I can confirm that the issue was that I was sourcing Noetic/Moveit1 in bashrc. 
By making sure to source only foxy I was able to complete the installation.