moveit/moveit2 #477: processCollisionObjectMove erroneously deletes existing subframe definitions

prajd3d, 2021-05-25:
### Description

processCollisionObjectMsg call erroneously deletes existing subframe definitions when called with a collision object msg with operation MOVE.

### Your environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch? main

### Steps to reproduce

Assume that a collision object is added to the planning scene via processCollisionObjectMsg (with the operation ADD). Also, assume that there are subframes in the collision object.

Now, after adding, I want to move the collision object in the planning scene.
So, I make a new collision object msg with a different mesh_poses. And, I set the operation field to MOVE.

After the processCollisionObjectMsg call, the subframes are no longer available. 

### Expected behaviour
The subframes should still be accessible after the move operation.

### Actual behaviour
The subframes are no longer available. (i.e., planning_scene->knowsFrameTransform("subframe") is false)

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

v4hn, 2021-05-25:
ping @felixvd . Does this affect [MoveIt](https://github.com/ros-planning/moveit) as well?

felixvd, 2021-05-26:
Yes, we left [the CollisionObjectMove function](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1801) unchanged because it was easier to do in the [object pose PR](https://github.com/ros-planning/moveit/pull/2037). That PR should fix the issue, it just hasn't been merged.

closed 2021-06-04 by prajd3d