moveit/moveit2 #490: How to set up a new arm from URDF for use with moveit2?

theunkn0wn1, 2021-06-09:
**Is your feature request related to a problem? Please describe.**
I am evaluating moveit2 for use with ROS2 Foxy as part of a club project. 
To this end, I am rather confused on how users are to import custom arm models for use with moveit2.
The documentation points [at outdated information](http://moveit2_tutorials.picknik.ai/doc/setup_assistant/setup_assistant_tutorial.html), and it seems this mentioned Setup Assistant is not available in moveit2 (#422).

**Describe the solution you'd like**
Clear and concise documentation describing how to set up Moveit2 using a custom robot model.

**Describe alternatives you've considered**
We are still early in the research phase, so I do not have a list of alternative libraries we are presently considering.

Using moveit1 via a ROS1 bridge is something we have dismissed, ROS2 is complicated enough without having to maintain a second equally complex environment for one component.

**Additional context**
We have our robot model in solidworks, though exporting it to URDF will not pose a problem. If there is a different format we should export into instead, any and all suggestions are welcome.

cgarry-vs, 2022-03-25:
It was pretty easy to do with MoveIt1. Just install ROS Noetic alongside your ROS2, source it, make a catkin workspace and create a package and add your urdf and mesh folders/files there (be sure to update your urdf to point to the correct mesh paths if youre copying it in from your ROS2 package) , and then just build the workspace. Then run the MSA and load the urdf and it should be able to find the meshes if you built the package properly. Then you can take the package generated from that and make a few changes so it works with MoveIt2.

MerdanBay, 2022-03-25:
> Then you can take the package generated from that and make a few changes so it works with MoveIt2.
can we find any tutorial or explanation about how to do those changes, @cgarry-vs

cgarry-vs, 2022-03-28:
@MerdanBay it depends on your package, but try to build and launch, and just fix the errors the logs complain about. I haven't gotten to that step yet, but other people have done it but not much info: https://rosindustrial.org/news/2020/4/29/using-moveit2-on-a-industrial-open-source-application

Let me know if you find anything general!

junaidamin, 2023-03-29:
Has anyone found a step-by-step tutorial on how to integrate a custom robot description into Moveit!2? It seems Moveit!2 doesn't have a setup assistant yet

okvik, 2023-03-29:
> Has anyone found a step-by-step tutorial on how to integrate a custom robot description into Moveit!2? It seems Moveit!2 doesn't have a setup assistant yet

Moveit Setup Assistant has been available for ROS 2 for a while now. It'll build along with MoveIt if you build from source or you can get it from binary packages (`ros-humble-moveit-setup-assistant`).

As for the tutorial, there's one at: https://moveit.picknik.ai/main/doc/examples/examples.html#integration-with-a-new-robot

okvik, 2023-03-29:
Hmm. Indeed, the tutorial seems to contain ROS 1 instructions. I could swear that was only the case for `humble` version, and the `main` branch that I linked above had the ROS 2 instructions... there's even an open pull request addressing that problem...  

It doesn't matter much though. Just substitute the `ros2` equivalent commands and the rest of the tool is both very similar to ROS 1 version and contains a decent amount of built-in help and instructions that you shouldn't have much trouble figuring out what you're supposed to do.

junaidamin, 2023-03-29:
Alright, I'll give it a shot. So far, the ROS 2 Moveit Setup Assistant seems very similar to the one shown in the ROS 1-based tutorial

Vedhamshbode, 2024-10-19:
Heyy!! Is the ROS2 Moveit Setup Assistant is stable?