moveit/moveit2 #496: Deal with ROS2 service & executors to get current robot states

amadeuszsz, 2021-06-14:
I am confused about provide ROS2 service with executors (as it mentioned in [moveit2 demo](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/move_group_interface/src/move_group_interface_tutorial.cpp)).

I already finished my service, unfortunately it works incorrectly. I mean, it is possible to move real robot, but I couldn't get current states, It works proper in demo mentioned above, so it exclude malfunctions on hardware side.

```
#include <moveit/move_group_interface/move_group_interface.h>
#include <rviz_visual_tools/rviz_visual_tools.hpp>
#include "custom_msgs/srv/move_arm.hpp"

class MoveRobot : public rclcpp::Node {
public:
    explicit MoveRobot(const rclcpp::NodeOptions &options)
            : Node("move_arm", options)
            , move_group_(static_cast<const SharedPtr>(this), "ur_manipulator")
    {
        parameters_client_ = std::make_shared<rclcpp::SyncParametersClient>(this, "move_group");
        service_ = this->create_service<custom_msgs::srv::MoveArm>(
                "move_arm_service", std::bind(&MoveRobot::move, this,
                                          std::placeholders::_1, std::placeholders::_2)
        );
    }

private:
    rclcpp::SyncParametersClient::SharedPtr parameters_client_;
    rclcpp::Service<custom_msgs::srv::MoveArm>::SharedPtr service_;
    moveit::planning_interface::MoveGroupInterface move_group_;

    void move(const std::shared_ptr<custom_msgs::srv::MoveArm::Request> request,
                     std::shared_ptr<custom_msgs::srv::MoveArm::Response> response)
    {
        auto pose = move_group_.getCurrentPose().pose;
        RCLCPP_INFO(this->get_logger(), "x: %f, y: %f, z: %f", pose.position.x, pose.position.y, pose.position.z);
        pose.position.x += request->x;
        pose.position.y += request->y;
        pose.position.z += request->z;
        moveit::planning_interface::MoveGroupInterface::Plan my_plan;
        bool success = (move_group_.plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
        if (success) RCLCPP_INFO(this->get_logger(), "\nPlanning finished successfully.");
        else RCLCPP_INFO(this->get_logger(), "\nPlanning failed.");
        move_group_.move();
        RCLCPP_INFO(this->get_logger(), "\nResponse: %d", response->success);
    }
};

int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);
    rclcpp::NodeOptions node_options;
    node_options.automatically_declare_parameters_from_overrides(true);
    auto move_group_node = std::make_shared<MoveRobot>(node_options);
    rclcpp::spin(move_group_node);
    rclcpp::shutdown();
    return 0;
}
```

I am able to execute `move_group_.move()` while calling service, but `move_group_.getCurrentPose()` returns zeros and some errors:
```
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1623665018.810613138] [move_group_interface]: Failed to fetch current robot state
```
I guess it may be necessary to use executors,  but I couldn't find any documentations, examples around it. I tried to create executor in initialization, directly in service function, main function and so on but usually it kills node immediately, in best scenario nothing changes. Is it expected behavior to make robot states unavailable with spin() but robot move execution possible? Is it necessary to use executors? How to use executor within services?

closed 2021-06-25 by amadeuszsz

YuLiHN, 2023-01-09:
I have the same problem too. Have you solved it?

ozangungortuhh, 2023-02-16:
Hello @amadeuszsz,
I am facing the exact same problem. Could you share your solution or your source code if you have found a solution.
Thanks a lot!

ozangungortuhh, 2023-02-16:
Thank you so much, it worked for me! @amadeuszsz 
Though I find it difficult to understand why we need two initialize our node two times?
Once with  `auto move_group_node = std::make_shared<MoveRobot>(node_options);` and also a class member `rclcpp::Node::SharedPtr node_;`
Thanks again for your help!

amadeuszsz, 2023-02-16:
> Thank you so much, it worked for me! @amadeuszsz Though I find it difficult to understand why we need two initialize our node two times? Once with `auto move_group_node = std::make_shared<MoveRobot>(node_options);` and also a class member `rclcpp::Node::SharedPtr node_;` Thanks again for your help!

To prevent deadlock. With current knowledge I guess I will solve it a little bit different but until it works you can stay with this approach. I suggest you to look up for **ros2 deadlock** and **ros2 executors** to understand it better.