moveit/moveit2 #499: build error: urdfom seems to be missing

91fabbai91, 2021-06-15:
I tried to build moveit2 from source as mentioned on the project page.
Therefore I used a clean installation of Ubuntu 20.04. After that I installed ros2 foxy with the debian package method.
The next step was to install/build moveit2, but I stops immediately when I start the colcon build with 
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

`Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Finished <<< moveit_common [0.39s]
Starting >>> controller_manager_msgs
--- stderr: srdfdom
CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:80 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:80 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:80 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:80 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:80 (ament_add_gtest)

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:35 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< srdfdom [0.61s, exited with code 1]
Aborted  <<< geometric_shapes [0.68s]
Aborted  <<< controller_manager_msgs [0.41s]
Aborted  <<< moveit_msgs [0.81s]

Summary: 1 package finished [1.17s]
  1 package failed: srdfdom
  3 packages aborted: controller_manager_msgs geometric_shapes moveit_msgs
  1 package had stderr output: srdfdom
  47 packages not processed
`

and to better understand I also paste in my vcs status below:

`.........
=== ./src/geometric_shapes (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/moveit2 (git) ===
Auf Branch main
Ihr Branch ist auf demselben Stand wie 'origin/main'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/moveit_msgs (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/moveit_resources (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/ros2_control (git) ===
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/ros2_controllers (git) ===
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/srdfdom (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/warehouse_ros (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
=== ./src/warehouse_ros_mongo (git) ===
Auf Branch ros2
Ihr Branch ist auf demselben Stand wie 'origin/ros2'.

nichts zu committen, Arbeitsverzeichnis unverändert
`
It is in german, but I think you can translate the messages.

Best regards,
Fabian

vatanaksoytezer, 2021-06-15:
Did you forget to run `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y` ? It seems the dependencies are not installed as intended. See https://moveit.ros.org/install-moveit2/source/

91fabbai91, 2021-06-15:
@vatanaksoytezer That's not the case. It says `#All required rosdepts installed successfully`

vatanaksoytezer, 2021-06-15:
I presume you have also sourced /opt/ros/foxy/setup.bash right?

91fabbai91, 2021-06-15:
yes, I did that and used `env | grep ROS` to look at the environment variables.
There are four variables
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=0 
ROS_DISTRO=foxy

Thanks for helping :)

chenguang3312, 2021-06-18:
I also encountered the same problem, delete the workspace and re-download and compile the solution。Maybe the system library is not up to date。
try：
`sudo apt update
sudo apt dist-upgrade
rosdep update
`

91fabbai91, 2021-06-21:
@chenguang3312 You are right. I had to delete the workspace and redownload it. I don't understand the issue, but it works so far.

So I will close this issue.

closed 2021-06-21 by 91fabbai91