moveit/moveit2 #504: Finish Windows port

ooeygui, 2021-06-17:
This Feature request is to finish the Windows port as started in https://github.com/ros-planning/moveit2/pull/238.

@lilustga and I (@ooeygui) will be working on this; this issue is to capture that workstream for tracking purposes.

Ace314159, 2021-06-21:
@ooeygui I have a [fork of MoveIt2](https://github.com/Ace314159/moveit2) that builds and properly runs for my team's usage. I didn't create a PR to merge it in since there was already an existing PR for a windows port, and I wanted to get the fixes for the other repos merged in first.

I'd love to combine our efforts and help with finishing the port.

ooeygui, 2021-06-21:
@Ace314159 Awesome! Would love to connect and discuss - can you add me to your network on linkedin? https://www.linkedin.com/in/louamadio/

Ace314159, 2021-06-22:
@ooeygui I've sent a request.

Ace314159, 2021-07-01:
I have created #530 which address this.

lilustga, 2021-10-09:
First Foxy release to include the Windows port fixes. https://github.com/ros-planning/moveit2/releases/tag/2.3.0

Now testing it on Windows!

Galactic testing soon as well

closed 2022-06-02 by henningkayser