moveit/moveit2 #509: run_moveit_cpp not work

chenguang3312, 2021-06-18:
terminal print:

`[run_moveit_cpp-4] [INFO] [1623981361.976500087] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[run_moveit_cpp-4] [INFO] [1623981361.976506719] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[run_moveit_cpp-4] [INFO] [1623981361.976509025] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[run_moveit_cpp-4] [INFO] [1623981361.976510891] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[run_moveit_cpp-4] [INFO] [1623981361.976512764] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[run_moveit_cpp-4] [WARN] [1623981361.977887269] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[run_moveit_cpp-4] [INFO] [1623981361.990606732] [moveit.plugins.moveit_simple_controller_manager]: Added FollowJointTrajectory controller for panda_arm_controller
[rviz2-3] [INFO] [1623981364.164479687] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[run_moveit_cpp-4] [WARN] [1623981366.991735300] [moveit.simple_controller_manager.gripper_controller_handle]: Waiting for panda_hand_controller/gripper_cmd to come up
[run_moveit_cpp-4] [WARN] [1623981371.992023830] [moveit.simple_controller_manager.gripper_controller_handle]: Waiting for panda_hand_controller/gripper_cmd to come up
[run_moveit_cpp-4] [ERROR] [1623981376.992440341] [moveit.simple_controller_manager.gripper_controller_handle]: Action client not connected: panda_hand_controller/gripper_cmd
[run_moveit_cpp-4] [INFO] [1623981376.997737643] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[run_moveit_cpp-4] [INFO] [1623981376.999149313] [moveit_ros.trajectory_execution_manager]: Trajectory execution is not managing controllers
[run_moveit_cpp-4] [INFO] [1623981376.999435586] [moveit_cpp_demo]: Initialize PlanningComponent
[run_moveit_cpp-4] [INFO] [1623981379.999865326] [moveit_cpp_demo]: Set goal
[run_moveit_cpp-4] [INFO] [1623981380.000071451] [moveit_cpp_demo]: Plan to goal
[run_moveit_cpp-4] [ERROR] [1623981380.000982309] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Workspace Bounds': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.002081718] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.002246350] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State In Collision': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.002632721] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.002789942] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State Bounds': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.003269463] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.003400566] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State In Collision': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.003754211] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.003899009] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Add Time Optimal Parameterization': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.003991963] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Workspace Bounds': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.004471685] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.004600133] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State In Collision': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.004967997] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.005118939] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State Bounds': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.005578350] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.005707353] [moveit.planning_request_adapter]: Exception caught executing *next* adapter 'Fix Start State In Collision': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.006043210] [moveit.planning_request_adapter]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': std::bad_alloc
[run_moveit_cpp-4] [ERROR] [1623981380.006214204] [moveit.ros_planning.planning_pipeline]: Exception caught: 'std::bad_alloc'
[run_moveit_cpp-4] [ERROR] [1623981380.006249208] [moveit.ros_planning_interface.planning_component]: Could not compute plan successfully
[run_moveit_cpp-4] [INFO] [1623981380.006359983] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'panda_arm'

`

MarqRazz, 2021-06-29:
@ChamzasKonstantinos I was running into this same issue the other day and it was because my source code was not up to date. Can you try pulling all of your source repos up to date?

JafarAbdi, 2021-06-30:
I'm pretty sure this's happening because you're using the wrong shared library, a clean build for your workspace should fix it

I'm going to close this issue, please feel free to reopen it if you still have the problem

closed 2021-06-30 by JafarAbdi