moveit/moveit2 #516: Pulling in new moveit2 changes breaks old workspace.

RyanDelGizzi, 2021-06-25:
### Description

Receiving build errors on `moveit_ros_visualization` after pulling in latest changes. The error comes from building the mocs and says that `QMetaObject::SuperData` has not been declared.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source Build
* Version: [vsctool repos file](https://github.com/ros-planning/moveit2/blob/fb94e78da7d9a16fee53ccc34fbfb2c7af2af5d0/moveit2.repos)

### Steps to reproduce
We have machines with a previous moveit workspace (perhaps from January or February).  That workspace was installed and worked fine.  If we do a `vcs pull` and update to all the newest repositories, we see the build failure described below.

### Expected behaviour
Build Succeeds

### Actual behaviour
Cannot build and receive the error in the linked gist

### Backtrace or Console output

[Build Output](https://gist.github.com/RyanDelGizzi/25470230bc9ee95774da4369273e856f)

### Workaround
The current workaround is to re-install ubuntu and do a fresh install of ROS2 and moveit2 and everything works.  This is not an ideal solution though, so if anyone has seen this or has any pointers for resolving the issue, it would be much appreciated.

vatanaksoytezer, 2021-06-27:
Hi @RyanDelGizzi thanks a lot for reporting! I've never seen this error and don't have much insight. However, I suspect it to be somewhat upstream since you mentioned you didn't update your workspace since January.

Rather then having a fresh Ubuntu install, simply calling `apt-get update && apt-get upgrade` to update your binary dependencies should fix your issue if my suspicion is correct. We already have a CI setup and running for Foxy, Rolling and Galactic that is run with every PR that compiles moveit2 in a fresh workspace and push and an automated docker build that pushes a pre-compiled moveit2 with a new ROS2 Foxy container to dockerhub to check our consistency.

Let me know if it helps so we can close the issue, or investigate more.

RyanDelGizzi, 2021-06-28:
Hi @vatanaksoytezer, 

Thank you for your attention to the issue. We discovered the failed build was caused by a conflicting build dependency with [ros scxml](https://github.com/swri-robotics/ros_scxml). I'll close the issue. :)

closed 2021-06-28 by RyanDelGizzi