moveit/moveit2 #522: Basic tutorial doesn't work

nkalupahana, 2021-06-30:
### Description

With the latest `moveit2` changes (42efad9), our team's MoveIt code suddenly stopped working. To see if it was an issue with our configuration, we tried the tutorials as well -- but they also seem to be broken. See "Steps to reproduce" for more.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20
* Source or Binary build? Source (42efad9)

### Steps to reproduce
- I created an empty Ubuntu 20 VM to start.
- Install `ros-foxy-desktop`
- [Follow the getting started instructions to install MoveIt2.](http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html)
- Run the MoveIt2 quickstart: `ros2 launch moveit2_tutorials demo.launch.py`

### Expected behaviour
You should get a screen that looks something like this (this is from my main Ubuntu VM, which has MoveIt2 at `a7d7798`:
[image]

### Actual behaviour
Instead, you get this:
[image]
In the "Context" tab of the MotionPlanning RViz plugin:
[image]

### Backtrace or Console output
Tutorial launch logs: https://gist.github.com/nkalupahana/09af17aa7f242bc73be5bd8495a7add5
My team's code launch logs: https://gist.github.com/nkalupahana/274d59bdaa42636109c00992cb124903
- Both the tutorial and my team's code work perfectly on MoveIt2 at `a7d7798`.

nkalupahana, 2021-06-30:
I've verified this with my old MoveIt2 installation and the latest build of `moveit2_tutorials` (tutorials still work) -- so this seems to be a problem with MoveIt itself and not a quirk of the tutorials. @henningkayser I'm pinging you on this directly because I believe this is a serious regression that should be addressed quickly, especially with Galactic release coming up. Hope that's okay :) and I'm open to provide additional testing details on anything.

JafarAbdi, 2021-06-30:
@nkalupahana Do you mind testing https://github.com/ros-planning/moveit2/pull/524 and see if it fixes the problem for you .?

closed 2021-06-30 by henningkayser

nkalupahana, 2021-06-30:
@JafarAbdi could you reopen this issue? This PR doesn't fix the problem -- I have the exact same issue. Y'all might want to run tutorials on `main` right now and see what happens for yourself. I'd be happy to test anything out too, my build machine is just somewhat slow.

https://gist.github.com/nkalupahana/08a294ad16dec6233c914910cf07c2f8

vatanaksoytezer, 2021-06-30:
@nkalupahana I just tested the tutorials and moveit2 (and all other upstreams) on main and can't seem to reproduce on Foxy. Do you mind cleaning your workspace, updating all the deps (by pulling and apt-get update / upgrade) and trying again and let us know if the issue persists?

JafarAbdi, 2021-06-30:
> @JafarAbdi could you reopen this issue? This PR doesn't fix the problem -- I have the exact same issue. Y'all might want to run tutorials on `main` right now and see what happens for yourself. I'd [...]

@nkalupahana Do you mind trying to start the `panda_hand_controller` controller  https://github.com/ros-planning/moveit2_tutorials/pull/95#issuecomment-871364195

nkalupahana, 2021-07-01:
I pulled up a blank VM and it looks like it's all working now -- not sure what was up with that :) #524 fixed it!