moveit/moveit2 #523: cannot build moveit core, not providing "FindTinyXML2.cmake"

QianBright, 2021-06-30:
Ubuntu 20.04 LTS
Linux ubuntu-ros 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
ROS2 foxy
moveit2

According to website http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html#install-ros2-and-colcon, I made the following error when I went to `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` step by step

```
Finished <<< velocity_controllers [4.00s]
Starting >>> ros2_controllers
--- stderr: moveit_core
CMake Error at /root/ws_moveit2/install/srdfdom/share/srdfdom/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "FindTinyXML2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TinyXML2",
  but CMake did not find one.

  Could not find a package configuration file provided by "TinyXML2" with any
  of the following names:

    TinyXML2Config.cmake
    tinyxml2-config.cmake

  Add the installation prefix of "TinyXML2" to CMAKE_PREFIX_PATH or set
  "TinyXML2_DIR" to a directory containing one of the above files.  If
  "TinyXML2" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /root/ws_moveit2/install/srdfdom/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:45 (find_package)

---
Failed   <<< moveit_core [5.01s, exited with code 1]
Aborted  <<< ros2_controllers [0.86s]

Summary: 32 packages finished [44.3s]
  1 package failed: moveit_core
  1 package aborted: ros2_controllers
  1 package had stderr output: moveit_core
  22 packages not processed
```

nkalupahana, 2021-06-30:
Did you run `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y`, like it says on the doc page? That should install everything -- I just did a clean install on an Ubuntu 20 machine about an hour ago and the everything built successfully.

QianBright, 2021-06-30:
yes..
```
root@ubuntu-ros:~/ws_moveit2/src# rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
#All required rosdeps installed successfully
```

JafarAbdi, 2021-06-30:
Can you make sure to update your srdfdom package .? this should be fixed with this [commit](https://github.com/ros-planning/srdfdom/commit/cadab16ca1ecf93e29ecb5b8e14505ccae080ebb)

QianBright, 2021-06-30:
When I update srdfdom, the result changes.

But it keeps showing `[Processing: moveit_core]` (when I do `colcon build` again) and doesn't move on to the next step.

JafarAbdi, 2021-06-30:
> When I update srdfdom, the result changes.
> 
> But it keeps showing `[Processing: moveit_core]` (when I do `colcon build` again) and doesn't move on to the next step.

Can you clean your workspace (remove build, install, and log directories) and try again .? also depending on your PC spec. moveit_core may take a while to compile, does your PC freeze .?

vatanaksoytezer, 2021-07-02:
I think the issue is not relevant anymore since it is fixed with https://github.com/ros-planning/srdfdom/commit/cadab16ca1ecf93e29ecb5b8e14505ccae080ebb. Feel free to re-open the issue if your problem persists.

closed 2021-07-02 by vatanaksoytezer