moveit/moveit2 #527: Panda KDL Singular test failed in a recent CI run on Foxy

vatanaksoytezer, 2021-06-30:
One of the recently added moveit_kinematics test has failed in the recent merge for some reason. The log doesn't show much info about the failure, but we may need to dig a little if this keeps failing. See https://github.com/ros-planning/moveit2/runs/2951969402 for log.

nkalupahana, 2021-06-30:
I think this might be related to #522 - also involves panda and KDL

vatanaksoytezer, 2021-06-30:
Thanks for the input! However, if that was the case I think we would get an assertion error on IK solving test. Here the test just dies and the other tests involving panda and kdl passes. I strongly think this is something else.

vatanaksoytezer, 2021-07-14:
Closing his since the flakiness doesn't seem to be here any more. I will open this back if I observe any CI failures related with these tests.

closed 2021-07-14 by vatanaksoytezer