moveit/moveit2 #535: Cannot load/reuse roadmap.graph for Persistent OMPL planners

tom-botbuilt, 2021-07-01:
### Description

I'm a new MoveIt2 user, and am trying to use the Persistent OMPL planners (i.e., PersistentLazyPRMstar and PersistentLazyPRM). I can generate the `roadmap.graph` file successfully, but trying to reuse it later causes an error message to display.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build, main branch

### Steps to reproduce
I'm following the "run_move_group" tutorial in moveit_demo_nodes. 
1. Add the PersistentLazyPRM planner configs to moveit_resources/panda_moveit_config/config/ompl_planning.yaml under `planner_configs`
```
PersistentLazyPRM: # use this to load a previously created roadmap  
    type: geometric::LazyPRM  
    multi_query_planning_enabled: 1  
    store_planner_data: 0  
    load_planner_data: 1  
    planner_data_path: /tmp/roadmap.graph  
PersistentLazyPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::LazyPRMstar  
    multi_query_planning_enabled: 1  
    store_planner_data: 1  
    load_planner_data: 0  
    planner_data_path: /tmp/roadmap.graph
```
Also add these under each section (planning group?) in ompl_planning.yaml: panda_arm, panda_arm_hand, panda_hand:
```
- PersistentLazyPRM
- PersistentLazyPRMstar
```
2. Make `PersistentLazyPRMstar` the `default_planner_config` for each planning group in ompl_planning.yaml. Run the demo to generate the `roadmap.graph` file at the end (or via CTRL-C after a planning step)
3. Make `PersistentLazyPRM` the `default_planner_config` for each planning group in ompl_planning.yaml and rerun the demo.
4. (error message appears)

### Expected behaviour
My understanding is that if the roadmap.graph is loaded correctly, the first planning step will say `Started planning with X states already in datastructure` where X is more than 2.

### Actual behaviour
The first planning step says `Starting planning with 2 states already in datastructure.` Just before that, there is an error message from planning_context_manager:

`[move_group-4] [ERROR] [1625166048.272018570] [moveit.ompl_planning.planning_context_manager]: Creating a 'panda_arm/panda_arm' planner from persistent data is not supported. Going to create a new instance.`

This occurs even though the existing roadmap.graph file is loaded correctly (I can verify the `numVertices` and `numEdges`) in planning_context_manager.cpp, line 178: `storage_.load(file_path.c_str(), data);`

I haven't seen any demos or documentation for successfully using Persistent planners; my impression was that simply changing the settings in `ompl_planning.yaml` was sufficient to switch between saving and loading a roadmap.graph. Any help would be appreciated!

tom-botbuilt, 2021-07-02:
EDIT: I fixed this issue by adding a function to planning_context_manager.cpp; basically re-added the functions that were removed since MoveIt1: 

```
template <>
inline ompl::base::Planner*
MultiQueryPlannerAllocator::allocatePersistentPlanner<ompl::geometric::LazyPRM>(const ob::PlannerData& data)
{
  return new og::LazyPRM(data);
```
(See [lines 194-224] (https://docs.ros.org/en/api/moveit_planners_ompl/html/planning__context__manager_8cpp_source.html))

v4hn, 2021-07-02:
So you found that [these specializations](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L194-L224) were removed in the ROS2 port and broke the functionality.
It looks like someone misunderstood the `TODO` at the top and removed the entire block instead of only the conditional...
I will create a corresponding pull-request to fix this.

As a rule of thumb: If you had to modify upstream source code to get an issue solved, the issue should not be closed before the change is actually upstream. Thanks for pointing it out and debugging!

PR #537 (2021-07-06): add persistent planner support back
These were accidentally removed during the port to ROS2 due to an ambiguous
`TODO: remove when melodic is no longer supported`

Closes #535 .
closed 2021-07-06 by JafarAbdi