moveit/moveit2 #536: Moving collision objects in RViz

christianlandgraf, 2021-07-02:
### Description

Strange behaviour when loading and moving scene objects.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source build, master

### Steps to reproduce
When loading any collision object into the scene, moving it through interactive marker control or the position boxes does not work as expected... (see video below). The position of the interactive marker also changes . Do you have any idea why this is happening? I couldn't find the solution or a similar issue so far...

Thanks a lot!

https://user-images.githubusercontent.com/28567122/124273665-976fab80-db40-11eb-87e2-a2319758c9f5.mp4

JafarAbdi, 2021-07-02:
Hi @christianlandgraf, I can't reproduce this, do you mind sharing more info .? what commit hash are you using .?

christianlandgraf, 2021-07-02:
Hi @JafarAbdi  I'm on the latest commit b8813c34d7576ea32e6549b96a87072d2cff8930 and using the recommended setup from `moveit.repos`. It's quite a strange behaviour...

I am using a custom description file ([this one](https://github.com/PickNikRobotics/UR10e_welding_demo/blob/master/ipa_demo_cell_description/urdf/ipa_ur_demo.urdf.xacro)). If the robot is linked to some other frame, e.g. a table, it seems as if the interactive marker for controlling the object and the object itself are somehow working in different coordinate frames. I couldn't fully track it down yet, it's just a first suspicion.

It's also possible that I have to align the [srdf](https://github.com/PickNikRobotics/UR10e_welding_demo/blob/master/ipa_moveit_config/srdf/ur_macro.srdf.xacro) and the coresponding urdf.

christianlandgraf, 2021-08-17:
The error seem to appear when the fixed frame in Rviz and the origin of the layout in MoveIt are different. I'm not sure if this is a common bug or a mistake in my Rviz configuration.
[image]

JafarAbdi, 2021-09-16:
@christianlandgraf Could you please test https://github.com/ros-planning/moveit2/pull/680 ? 

Thanks!

christianlandgraf, 2021-09-16:
@JafarAbdi It works for me now, thanks a lot for fixing it!

closed 2022-02-07 by henningkayser