moveit/moveit2 #541: joint limit loading and overwriting behavior

brianyany, 2021-07-07:
A few things: 
1. joint_limits.yaml not loaded in the demo apps and not clear how exactly would one go about loading it. 
2. When joint_limits.yaml is not loaded, in robot_model_loader.cpp, the configure function would attempts to load the limits but fail (which means there's no pos, vel, or acc limit). Then the call at line 226 to setVariableBounds would overwrite the joint limits read from the urdf file. This behavior seems strange: shouldn't the joint_model retain the limits read from the urdf file? 
3. When joint_limits are not properly loaded and close to singularity, position/velocity value would explode..

JafarAbdi, 2021-07-08:
> 1. joint_limits.yaml not loaded in the demo apps and not clear how exactly would one go about loading it.

This [PR](https://github.com/ros-planning/moveit2/pull/544) add how to load the joint limits

> 2. When joint_limits.yaml is not loaded, in robot_model_loader.cpp, the configure function would attempts to load the limits but fail (which means there's no pos, vel, or acc limit). Then the call [...]

The joint_limit is initially set to the values from the urdf in [L125](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/robot_model_loader/src/robot_model_loader.cpp#L125), and will only be overridden if `joint_limits.yaml` is loaded

> 3. When joint_limits are not properly loaded and close to singularity, position/velocity value would explode..

Do you mind sharing a reproducible example .?

JafarAbdi, 2021-07-08:
> Have to disagree with the point you made about 2 since I am observing the limit values got overwritten without loading joint_limits.yaml. If you add a print for bounds.velocity_bounded_ at line 765 of servo_calcs.cpp. You will notice that bounds.velocity_bounded_ evaluates to false. However, if you comment out the call to setVariableBounds at line 225 of robot_model_loader.cpp, bounds.velocity_bounded_ evaluates to true in servo_calcs.cpp. This means the joint limits from urdf got overwritten by the unsuccessful attempt to load limit values from joint_limits.yaml. (Running one of the servo demos)

You're right, can you test this [PR](https://github.com/ros-planning/moveit2/pull/546) and see if it fixes the bug for you .?

brianyany, 2021-07-08:
> You're right, can you test this [PR](https://github.com/ros-planning/moveit2/pull/546) and see if it fixes the bug for you .?

yep! this PR resolved the joint limits overwritten issue! Thank you for the quick fix.

JafarAbdi, 2021-09-28:
Fixed with the mentioned PRs

closed 2021-09-28 by JafarAbdi