moveit/moveit2 #543: Failed to detect successful installation of [ros-galactic-random-numbers] and others

NebuDa, 2021-07-08:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Galactic
* OS Version: e.g. Ubuntu 20.04
* Binary build
* via https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html

### Steps to reproduce
Hi there,
I have installed ROS2 Galactic form Debian packages in Windows Subsystem for Linux2 and it runs great. Now I have tried to install MoveIt2 with https://moveit.ros.org/install-moveit2/source/ and got stuck at the rosdep install Command. For output see down below. This causes that the colcon build process cannot find the packages listed below.

###  Console output

flo@DESKTOP-L4I9CIG:~/ws_moveit2/src$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
executing command [sudo -H apt-get install -y ros-galactic-random-numbers]
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-galactic-random-numbers
executing command [sudo -H apt-get install -y ros-galactic-object-recognition-msgs]
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-galactic-object-recognition-msgs
executing command [sudo -H apt-get install -y ros-galactic-control-toolbox]
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ros-galactic-control-toolbox
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-galactic-random-numbers] failed
  apt: command [sudo -H apt-get install -y ros-galactic-object-recognition-msgs] failed
  apt: command [sudo -H apt-get install -y ros-galactic-control-toolbox] failed
  apt: Failed to detect successful installation of [ros-galactic-random-numbers]
  apt: Failed to detect successful installation of [ros-galactic-object-recognition-msgs]
  apt: Failed to detect successful installation of [ros-galactic-control-toolbox]

JafarAbdi, 2021-07-08:
For galactic you need to use https://github.com/ros-planning/moveit2/blob/main/moveit2_galactic.repos rather than the one shown in [here](https://moveit.ros.org/install-moveit2/source/#download-source-code)

I opened a PR to fix the tutorial https://github.com/ros-planning/moveit.ros.org/pull/617

closed 2021-07-08 by JafarAbdi

NebuDa, 2021-07-08:
Thanks a lot. It works.