moveit/moveit2 #547: Add command line interface to moveit

JafarAbdi, 2021-07-08:
Having CLI similar to [ros2controlcli](https://github.com/ros-controls/ros2_control/tree/master/ros2controlcli) that could be used for debugging would be very helpful for users, for example, wrapping some of the following functionalities https://github.com/ros-planning/moveit2/tree/main/moveit_ros/planning/planning_components_tools

v4hn, 2021-07-09:
I believe the `moveit_commander_cmdline.py` is meant to serve this purpose. Or do you have something different in mind?
Either way I agree that the commander cmd does not provide enough support for debugging.

JafarAbdi, 2021-07-16:
The problem we have here is not having a global parameter server in ROS2, which mean the user have to pass all the parameters used for move_group ([example](https://github.com/ros-planning/moveit2/blob/main/moveit_demo_nodes/run_move_group/launch/run_move_group.launch.py)) to `moveit_commander_cmdline.py` which is a big overhead for users, so my solution is to add a new move group capability that let the user query all these info