moveit/moveit2 #548: Contribution to the development of the new Moveit2 Setup Assistant

Afif-Swaidan, 2021-07-08:
Hello it's Afif Swaidan from Spexal Robotics. We are a Robotics company that is mainly focused on logistics robotics. We already did several projects that utilized ROS navigation for mobile robots and ROS manipulation (Moveit) for robotic arms. We are already migrating to ROS2 as well and we are interested to contribute in the process of migrating MoveIt2 to ROS2.

Since the majority of MoveIt2 Core features are already done and the only missing feature now is the Moveit Setup Assistant for Moveit2 to setup new robotic arms through a graphical user interface. We already modified the old Moveit Setup Assistant to start working with ROS2.

That being said, we would love to contribute in the development of the new Moveit Setup Assistant especially that we already did some work in this process. We are ready to be added to the weekly moveit2 stand-up at anytime!

We are a team of 3 ROS developers:
- Afif Swaidan, Robotics Engineer (afif.swaidan@spexal.com)
- Mario Korban, Software Engineer (mario.korban@spexal.com)
- Jad Helou, Mechanical Engineer (jad.helou@spexal.com)

davetcoleman, 2021-07-10:
We'd love your support in the migration! @henningkayser is the expert in the remaining needs here.

DLu, 2021-07-12:
The main issue tracking the migration is https://github.com/ros-planning/moveit2/issues/313

Afif-Swaidan, 2021-07-12:
@henningkayser Well received,
Thanks!

closed 2021-07-12 by Afif-Swaidan