moveit/moveit2 #557: CI is broken for rolling

JafarAbdi, 2021-07-16:
### Description

As you can see https://github.com/ros-planning/moveit2/runs/3084812617#step:6:944, CI is failing for rolling with the following error

```
  --- stderr: controller_manager
  CMake Error at CMakeLists.txt:17 (find_package):
    Found package configuration file:
  
      /home/runner/work/moveit2/moveit2/.work/upstream_ws/install/controller_manager_msgs/share/controller_manager_msgs/cmake/controller_manager_msgsConfig.cmake
  
    but it set controller_manager_msgs_FOUND to FALSE so package
    "controller_manager_msgs" is considered to be NOT FOUND.  Reason given by
    package:
  
    The following imported targets are referenced, but are missing:
    controller_manager_msgs::controller_manager_msgs__rosidl_typesupport_introspection_cpp
  
  
  
  
  ---
  Failed   <<< controller_manager [2.07s, exited with code 1]
```

JafarAbdi, 2021-07-16:
Should be fixed by https://github.com/ros2/rosidl/issues/602

vatanaksoytezer, 2021-07-19:
We can add this to repos until sync to avoid CI failures @JafarAbdi @henningkayser @tylerjw

vatanaksoytezer, 2021-07-20:
They had a fix sync yesterday. This will fix our CI and we can close this issue.

nkalupahana, 2021-07-26:
Our CI still doesn't seem to be fixed (see rolling CI on #530, #555, any of the new PRs such as #571, etc.). If this is still an issue on ROS's side, I think this and related issues should probably be re-opened.

vatanaksoytezer, 2021-07-27:
To clarify the status of Rolling, CI is actually fixed with the sync and then it got broke again with another reason.

The earlier rolling CI fail was related with: https://github.com/ros2/rosidl/issues/602

Now it is: https://github.com/osrf/infrastructure/issues/5

No need for another issue though, we can keep using this one.

tylerjw, 2021-08-11:
I think this is now fixed, re-open if it is not.

closed 2021-08-11 by tylerjw