moveit/moveit2 #559: 1 package failed: rviz_visual_tools

lvhao54, 2021-07-20:
### Description

1 package failed: rviz_visual_tools.

### Your environment
* ROS2 Distro: [foxy]
* OS Version: e.g. Ubuntu 20.04
* Source build
* If source, which branch
* Create A Colcon Workspace and Download MoveIt2 Source
These tutorials rely on the master branch of MoveIt2, which requires a build from source. You will need to have a colcon workspace setup:

mkdir -p ~/ws_moveit2/src
cd ~/ws_moveit2/src

git clone https://github.com/ros-planning/moveit2.git
git clone https://github.com/ros-planning/moveit2_tutorials.git
vcs import < moveit2/moveit2.repos
vcs import < moveit2_tutorials/moveit2_tutorials.repos
Build your Colcon Workspace
The following will install from Debian any package dependencies not already in your workspace:

rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
The next command will configure your colcon workspace:

cd ~/ws_moveit2
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
build success

### Actual behaviour
--- stderr: rviz_visual_tools
In file included from /home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/mocs_compilation.cpp:2:
/home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/5KG5KIFYQQ/moc_key_tool.cpp:67:18: error: ‘QMetaObject::SuperData’ has not been declared
   67 |     QMetaObject::SuperData::link<rviz_common::Tool::staticMetaObject>(),
      |                  ^~~~~~~~~
/home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/5KG5KIFYQQ/moc_key_tool.cpp:67:71: error: expected primary-expression before ‘)’ token
   67 |     QMetaObject::SuperData::link<rviz_common::Tool::staticMetaObject>(),
      |                                                                       ^
In file included from /home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/mocs_compilation.cpp:3:
/home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/5KG5KIFYQQ/moc_rviz_visual_tools_gui.cpp:94:18: error: ‘QMetaObject::SuperData’ has not been declared
   94 |     QMetaObject::SuperData::link<rviz_common::Panel::staticMetaObject>(),
      |                  ^~~~~~~~~
/home/amber/ws_moveit2/build/rviz_visual_tools/rviz_visual_tools_gui_autogen/5KG5KIFYQQ/moc_rviz_visual_tools_gui.cpp:94:72: error: expected primary-expression before ‘)’ token
   94 |     QMetaObject::SuperData::link<rviz_common::Panel::staticMetaObject>(),
      |                                                                        ^
make[2]: *** [CMakeFiles/rviz_visual_tools_gui.dir/build.make:63: CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_autogen/mocs_compilation.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:320: CMakeFiles/rviz_visual_tools_gui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_visual_tools [39.6s, exited with code 2]
Aborted  <<< controller_manager [6.61s]
Aborted  <<< moveit_core [3min 37s]

Summary: 17 packages finished [3min 54s]
  1 package failed: rviz_visual_tools
  2 packages aborted: controller_manager moveit_core
  1 package had stderr output: rviz_visual_tools
  36 packages not processed

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

vatanaksoytezer, 2021-07-20:
Hello @lvhao54, thanks for reporting!

Neither I or the CI can't reproduce this. Can you send the full log of your failure? Do you have a sourced ROS workspace (`source /opt/ros/foxy/setup.bash`)?

lvhao54, 2021-07-21:
[latest_build.zip](https://github.com/ros-planning/moveit2/files/6852371/latest_build.zip)

lvhao54, 2021-07-21:
> 你好@lvhao54，感谢报告！
> 
> 我或 CI 都无法重现这一点。你能发送完整的失败日志吗？你有源 ROS 工作区 ( `source /opt/ros/foxy/setup.bash`) 吗？

[latest_build.zip](https://github.com/ros-planning/moveit2/files/6852373/latest_build.zip)

lvhao54, 2021-07-21:
> Hello @lvhao54, thanks for reporting!
> 
> Neither I or the CI can't reproduce this. Can you send the full log of your failure? Do you have a sourced ROS workspace (`source /opt/ros/foxy/setup.bash`)?

I exec the sourced ROS Workspace (`source /opt/ros/foxy/setup.bash`).
I wonder if it is related to my installation of QT5.15.2.

lvhao54, 2021-07-21:
> Hello @lvhao54, thanks for reporting!
> 
> Neither I or the CI can't reproduce this. Can you send the full log of your failure? Do you have a sourced ROS workspace (`source /opt/ros/foxy/setup.bash`)?

closed 2021-07-22 by lvhao54