moveit/moveit2 #574: Parse jerk limits from joint_limits.yaml and add jerk limits to Joint Model

AndyZe, 2021-07-28:
Let's give MoveIt the functionality of processing jerk limits.

tlpss, 2021-09-25:
Hi @AndyZe, could you provide some details on this? And maybe an estimation of the required implementation time? 

I'm looking to make some first contributions to Moveit.

AndyZe, 2021-09-26:
Shoot, I already implemented it myself in #683. Sorry I forgot to link it with this issue.

You could always give that PR a code review. We all appreciate that and it's a good way to get started.

If you're looking for something more ambitious, like 20 hours of work, I have some ideas to improve this recent [trajectory smoothing PR](https://github.com/ros-planning/moveit2/pull/571).

What are your interests/skills and we could help you find something, I'm sure. Motion planning? Controls? GUI?

AndyZe, 2021-09-26:
Writing tests is always a good way to get started, too. They are really important to catch regressions and occasionally catch existing bugs.

tlpss, 2021-09-27:
@AndyZe writing some tests seems like a good idea, if you have something in mind feel free to provide some directions. 
I'm rather new to Moveit but my interest would be mostly in control or planning.

AndyZe, 2021-09-27:
A test for this PR would be great:   https://github.com/ros-planning/moveit2/pull/707

Basically you could just browse through the code and find a package that is short on tests.

Lots of tutorials still need to be ported, too. Here's a good example that (I think...) would not be too hard to port:  http://moveit2_tutorials.picknik.ai/doc/subframes/subframes_tutorial.html

AndyZe, 2021-09-28:
@tlpss Here's a very easy issue to get started with, if you want   ;)

https://github.com/ros-planning/moveit2_tutorials/issues/122

PR #683 (2022-01-02): Add jerk to the robot model
Now that we have some MoveIt capabilities that can process jerk limits, I think we should add jerk limits to the robot model. This simply adds the jerk parameters to joint_limits.yaml and adds the parsing code to robot_model_loader.cpp.

Fixes #574
closed 2022-01-02 by AndyZe