moveit/moveit2 #588: able to build moveit on foxy: resource_retriever/retriever.hpp: No such file or directory

odbee, 2021-08-02:
### Description

whenever i try to build the current version of moveit i get an error during the install of geometric_shapes and the build fails.
of course i did `rosdep install -r --from-paths . --ignore-src --rosdistro foxy -y
` before

### My environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* built from source, master branch

### Steps to reproduce
simply running 
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`
after following the tutorial on http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html#create-a-colcon-workspace-and-download-moveit2-source

### Actual behaviour
the build fails

### Backtrace or Console output

```zsh
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> controller_manager_msgs
Starting >>> warehouse_ros
Starting >>> moveit_resources_panda_description
Starting >>> ros2_control_test_assets
Finished <<< moveit_common [2.44s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_panda_description [2.46s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< ros2_control_test_assets [2.53s]
Starting >>> hardware_interface
Finished <<< moveit_resources_fanuc_description [7.99s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_moveit_config [8.17s]
Starting >>> moveit_resources_pr2_description
[Processing: controller_manager_msgs, geometric_shapes, hardware_interface, moveit_msgs, moveit_resources_fanuc_moveit_config, moveit_resources_pr2_description, srdfdom, warehouse_ros]
Finished <<< moveit_resources_fanuc_moveit_config [30.6s]
Starting >>> rviz_visual_tools
--- stderr: geometric_shapes
/home/ed/workspace/moveit2_ws/src/geometric_shapes/src/mesh_operations.cpp:41:10: fatal error: resource_retriever/retriever.hpp: No such file or directory
   41 | #include <resource_retriever/retriever.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/geometric_shapes.dir/build.make:102: CMakeFiles/geometric_shapes.dir/src/mesh_operations.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/geometric_shapes.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< geometric_shapes [41.1s, exited with code 2]
Aborted  <<< rviz_visual_tools [0.03s]
Aborted  <<< moveit_resources_pr2_description [30.4s]
Aborted  <<< warehouse_ros [47.8s]
Aborted  <<< srdfdom [48.8s]
Aborted  <<< controller_manager_msgs [58.2s]
Aborted  <<< hardware_interface [1min 30s]
Aborted  <<< moveit_msgs [3min 27s]

Summary: 6 packages finished [3min 28s]
  1 package failed: geometric_shapes
  7 packages aborted: controller_manager_msgs hardware_interface moveit_msgs moveit_resources_pr2_description rviz_visual_tools srdfdom warehouse_ros
  1 package had stderr output: geometric_shapes
  42 packages not processed
```

tylerjw, 2021-08-02:
There is now a [foxy branch of moveit2](https://github.com/ros-planning/moveit2/tree/foxy).  The hpp file you see in your build output only exists on rolling/galactic and the main branch of moveit2 only supports rolling and galactic now.  I'm sorry if this caused you trouble, the branch was made last week.

tylerjw, 2021-08-02:
I created PR here to fix the tutorials: https://github.com/ros-planning/moveit2_tutorials/pull/100

odbee, 2021-08-03:
hey @tylerjw thannks for your quick reply but this doesnt fix it unfortunately. i think the problem lies in the 'ros2' branch of [geometric shapes](https://github.com/ros-planning/geometric_shapes)  where someone just changed the header file from retriever.h to retriever.hpp as stated here:

https://github.com/ros-planning/geometric_shapes/commit/42a4a0f178749eeb17486bf2263d72a057e8f2f2#diff-95b01efdb0b85e4290ef07fe7f043b29f099861c0d98d7df66023a322069e80a

this breaks the foxy build since its still running on the [foxy branch](https://github.com/ros/resource_retriever/tree/foxy/resource_retriever/include/resource_retriever) of resource retriever with the .h file instead of the .hpp file on the [ros2 branch](https://github.com/ros/resource_retriever/tree/ros2/resource_retriever/include/resource_retriever).

i fixed my issue by changing the geometric_shapes file undoing the change from the commit of Drojas251 but a general fix would be to change the branches of the [moveit2.repos](https://github.com/ros-planning/moveit2/blob/foxy/moveit2.repos) file for the foxy branch. unfortunately i dont have the time to try out all options to check if theres any depedencies that would break but please keep the issue open until it is resolved.

JafarAbdi, 2021-08-03:
> hey @tylerjw thannks for your quick reply but this doesnt fix it unfortunately. i think the problem lies in the 'ros2' branch of [geometric shapes](https://github.com/ros-planning/geometric_shapes) where someone just changed the header file from retriever.h to retriever.hpp as stated here:
> 
> [ros-planning/geometric_shapes@42a4a0f#diff-95b01efdb0b85e4290ef07fe7f043b29f099861c0d98d7df66023a322069e80a](https://github.com/ros-planning/geometric_shapes/commit/42a4a0f178749eeb17486bf2263d72a057e8f2f2#diff-95b01efdb0b85e4290ef07fe7f043b29f099861c0d98d7df66023a322069e80a)
> 
> this breaks the foxy build since its still running on the [foxy branch](https://github.com/ros/resource_retriever/tree/foxy/resource_retriever/include/resource_retriever) of resource retriever with the .h file instead of the .hpp file on the [ros2 branch](https://github.com/ros/resource_retriever/tree/ros2/resource_retriever/include/resource_retriever).
> 
> i fixed my issue by changing the geometric_shapes file locally and undoing the change from the commit of Drojas251 but a general fix would be to change the branches of the [moveit2.repos](https://github.com/ros-planning/moveit2/blob/foxy/moveit2.repos) file for the foxy branch. unfortunately i dont have the time to try out all options to check if theres any depedencies that would break but please keep the issue open until it is resolved.

Should  be fixed in https://github.com/ros-planning/moveit2/pull/589, @odbee could you please review it .?

odbee, 2021-08-03:
hey @JafarAbdi unfortunately I dont think i have permission to review, but it doesnt work. when changing the geometric_shapes branch to foxy it collides with srfdom and probably all other packages in the moveit2.repos that dont have a foxy branch because they seem to be behind

tylerjw, 2021-08-03:
Do you have local changes to srfdom?  For me at least when working in foxy I'm able to work with just moviet2 checked out and the rest of the stuff installed with rosdep?

Can you post the error you get from building srdfdom with the foxy branch of geometric_shapes?

Thank you for helping us figure out these issues.  Many of us are working on rolling because in ros2 itself there are a bunch of bugs in foxy that are just fixed on rolling.

odbee, 2021-08-04:
for sure, here you go:
```zsh
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> controller_manager_msgs
Starting >>> moveit_common
Starting >>> ros2_control_test_assets
Starting >>> warehouse_ros
Starting >>> moveit_resources_panda_description
Finished <<< moveit_common [1.30s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_panda_description [1.32s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< ros2_control_test_assets [1.43s]
Starting >>> hardware_interface
--- stderr: srdfdom
CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_nl_NL.UTF-8" links to target
  "urdfdom_headers::urdfdom_headers" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:84 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_nl_NL.UTF-8" links to target
  "urdfdom_headers::urdfdom_headers" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:84 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_nl_NL.UTF-8" links to target
  "urdfdom_headers::urdfdom_headers" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:84 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_nl_NL.UTF-8" links to target
  "urdfdom_headers::urdfdom_headers" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:84 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_nl_NL.UTF-8" links to target
  "urdfdom_headers::urdfdom_headers" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:84 (ament_add_gtest)

CMake Error at /opt/ros/foxy/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_cpp_C" links to target "urdfdom_headers::urdfdom_headers" but
  the target was not found.  Perhaps a find_package() call is missing for an
[... 61 lines cut ...]

CMake Error at CMakeLists.txt:34 (add_library):
  Target "srdfdom" links to target "urdfdom_headers::urdfdom_headers" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< srdfdom [3.57s, exited with code 1]
Aborted  <<< moveit_resources_panda_moveit_config [2.27s]
Aborted  <<< moveit_resources_fanuc_description [2.33s]
Aborted  <<< warehouse_ros [3.70s]
Aborted  <<< geometric_shapes [3.76s]
Aborted  <<< hardware_interface [3.18s]
Aborted  <<< moveit_msgs [6.18s]
Aborted  <<< controller_manager_msgs [17.6s]

Summary: 3 packages finished [18.1s]
  1 package failed: srdfdom
  7 packages aborted: controller_manager_msgs geometric_shapes hardware_interface moveit_msgs moveit_resources_fanuc_description moveit_resources_panda_moveit_config warehouse_ros
  1 package had stderr output: srdfdom
  43 packages not processed
```

as i said, it seems like the foxy branch of geometric_shapes is outdated and the best way to fix it would be to update the foxy branch of geometric_shapes to ros2 branch and just change the resource retriever include.

JafarAbdi, 2021-08-04:
> as i said, it seems like the foxy branch of geometric_shapes is outdated and the best way to fix it would be to update the foxy branch of geometric_shapes to ros2 branch and just change the resource retriever include.

The CMake config step is failing for `srdfdom`, how is this related to `geometric_shapes` .?

@odbee Can you test this [PR](https://github.com/ros-planning/srdfdom/pull/96) to see if it fixes the srdfdom failure .?

JafarAbdi, 2021-08-04:
Also, can you try a clean build of the workspace .? look like your error is similar to https://github.com/ros-planning/moveit2/issues/499

vatanaksoytezer, 2021-08-04:
> Also, can you try a clean build of the workspace .? look like your error is similar to #499

I think that is the issue. Just clean your build and install of your workspace and build making sure to use the same branches in moveit2.repos (in moveit2 foxy branch).

tylerjw, 2021-08-04:
> foxy branch of geometric_shapes is outdated

Afaik the only real difference between the foxy and ros2 branch of geometric_shapes is the resource retriever include.

JafarAbdi, 2021-09-28:
Please feel free to re-open this if you still have the problem

closed 2021-09-28 by JafarAbdi