moveit/moveit2 #597: missing run dependencies for servo demos

tylerjw, 2021-08-10:
### Description

This is some of the output of running: `ros2 launch moveit_servo servo_teleop.launch.py`.

```
[ros2_control_node-2] [INFO] [1628569196.077672859] [controller_manager]: Loading controller 'joint_state_controller'
[ros2_control_node-2] [ERROR] [1628569196.077743017] [controller_manager]: Available classes:
[ros2_control_node-2] [ERROR] [1628569196.077755482] [controller_manager]: controller_manager/test_controller
[ros2_control_node-2] [ERROR] [1628569196.077762294] [controller_manager]: controller_manager/test_controller_failed_init
[ros2_control_node-2] [ERROR] [1628569196.077768948] [controller_manager]: controller_manager/test_controller_with_interfaces
[ros2_control_node-2] [ERROR] [1628569196.077775506] [controller_manager]: Loader for controller 'joint_state_controller' not found.
[ros2_control_node-2] [INFO] [1628569196.083868697] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2_control_node-2] [ERROR] [1628569196.083905842] [controller_manager]: Available classes:
[ros2_control_node-2] [ERROR] [1628569196.083915106] [controller_manager]: controller_manager/test_controller
[ros2_control_node-2] [ERROR] [1628569196.083921690] [controller_manager]: controller_manager/test_controller_failed_init
[ros2_control_node-2] [ERROR] [1628569196.083927818] [controller_manager]: controller_manager/test_controller_with_interfaces
[ros2_control_node-2] [ERROR] [1628569196.083934001] [controller_manager]: Loader for controller 'panda_arm_controller' not found.
[ERROR] [ros2 run controller_manager spawner.py joint_state_controller-5]: process has died [pid 135003, exit code 1, cmd 'ros2 run controller_manager spawner.py joint_state_controller'].
[ERROR] [ros2 run controller_manager spawner.py panda_arm_controller-4]: process has died [pid 135001, exit code 1, cmd 'ros2 run controller_manager spawner.py panda_arm_controller'].
```

What are the run dependencies we are missing?  Have these been released for rolling/galactic?

tylerjw, 2021-08-12:
I just realized that `servo_teleop.launch.py` is a file only I have locally :facepalm: it is a copy of the file from the tutorial for my own testing.

AndyZe, 2021-08-21:
The error with loading `panda_arm_controller` can be fixed by `sudo apt install ros-rolling-joint-trajectory-controller`.

For joint state controller: `sudo apt install ros-rolling-joint-state-broadcaster`

It looks like Vatan's PR #603 fixes the joint-state-controller. I believe we still ought to add joint-trajectory-controller

AndyZe, 2021-08-21:
I guess it works for now because the dependency is brought in by a different package.

PR #636 (2021-08-21): Add gripper and traj control packages as run dependencies
Fixes #597 

Similar to #603 except it missed these runtime dependencies. I think it worked anyway because they were brought in by a different package (transitive dependency)
closed 2021-08-21 by tylerjw