moveit/moveit2 #601: CMake Error - get_target_property called with incorrect number of arguments

lydiazoghbi, 2021-08-10:
Hi! I'm having the same problem as in #446 with my installation. Basically some packages won't build because of some "get_target_property called with incorrect number of arguments". Using Ubuntu 20.04, Foxy - I'm on branch Foxy (as recommended per the UR driver developers). I'm pasting the error output from the terminal for your reference. I tried the suggestions posted on thread #446 but it didn't really do much. I would appreciate any feedback on the issue. Thanks!

```
Starting >>> moveit_simple_controller_manager
--- stderr: moveit_ros_occupancy_map_monitor
CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
[... 294 lines cut ...]
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

CMake Error at /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/imerse/Documents/summer21/ros2_ur_driver/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

---
Failed   <<< moveit_ros_occupancy_map_monitor [2.93s, exited with code 1]
Aborted  <<< moveit_simple_controller_manager [3.18s]
```

tylerjw, 2021-08-10:
What branch of moveit are you using (there is a foxy branch of moveit and many of our upstream repos you might have in your workspace as the main and ros2 branches are now for rolling and galactic)?  Do you mind running `vcs branch` in your src directory and posting the output here?

That being said I looked at the issue you linked to and all I could gather is that person found it was some cmake issue.  What command are you running when you build?  What version of cmake are you using?  Is this on MacOS like the original person who reported this?

lydiazoghbi, 2021-08-11:
Hey! I'm using the foxy branch, `vcs branch` returns foxy. 

Yeah I linked the previous issue because I was getting the same exact error output, but I don't think it's the same cmake issue on my end. In other words I didn't build with `--cmake-force-configure`. 

I am building using `colcon build`. Building with `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release' returns the same exact error as well. I have cmake version 3.16.3, and no it's on a natively installed linux, ubuntu 20.

lydiazoghbi, 2021-08-11:
I just created a brand new workspace and followed your suggestions, but still get the same error..

lydiazoghbi, 2021-08-11:
Update: I had two ros versions on my system - noetic and foxy. Purging noetic solved the issue. Although I sourced foxy in my workspace, apparently the build was still linking to an .so file in /opt/ros/noetic for some reason (got an error message). Had to completely uninstall noetic and related dependencies. It builds fine now. Thanks for the help!

closed 2021-08-11 by lydiazoghbi

tylerjw, 2021-08-11:
Thank you for documenting that.  I suspect you had a line in your .bashrc that sources the noetic install.  One of the new "features" of ros2 is that it does not try to switch wich version of ros you are working with when you source the ros install.  That means that if you had previously sourced your noetic install in your bashrc and then sourced your foxy install before you built, noetic is still in your enviroment.

abake48, 2021-08-24:
I happened to run into this issue today while Foxy was the only version installed on my machine. Performing a complete rebuild by removing the build, log, and install directories from my workspace seemed to do the trick for me. Unsure what caused the issue initially though.