moveit/moveit2 #607: Galactic main pool missing debian files

Briancbn, 2021-08-12:
### Description

The following packages doesn't seemed to have a `.deb` file in the the main pool. [example](http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-galactic-moveit-ros-planning-interface/) vs [testing-pool](http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-galactic-moveit-ros-planning-interface/). 
- moveit-ros-planning-interface
- moveit-ros-move-group
- moveit-kinematics
- moveit-planners-ompl
- moveit-ros-visualization
- moveit-servo

As these packages are listed in [rosdistro](https://github.com/ros/rosdistro/blob/dafdd5af1a757b248b118ba70ec4776badb8c4f1/galactic/distribution.yaml#L1538-L1544), `rosdep` is able to resolve these dependencies automatically, but could not install them.

Would this problem be resolved simply after the next sync?

### Your environment
* ROS Distro: Galactic
* OS Version: e.g. Ubuntu 20.04
* Binary build
* 2.2.0

### Steps to reproduce
Run the install command.
```bash
sudo apt-get install -y ros-galactic-moveit-planners-ompl
```

### Expected behaviour
Install without problem.

### Actual behaviour
Cannot install
```bash
E: Unable to locate package ros-galactic-moveit-servo
```

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

NA

schornakj, 2021-08-12:
+1, I'm running into this as well in CI for Galactic. I get errors like this:

```
executing command [apt-get install -y -qq ros-galactic-moveit-ros-planning-interface]
  E: Unable to locate package ros-galactic-moveit-ros-planning-interface
  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y -qq ros-galactic-moveit-ros-planning-interface] failed
```

henningkayser, 2021-08-12:
@Briancbn you're right, we are simply waiting for the next Galactic sync. Until then, you'll have to stick with the testing version, unfortunately. I'll close this issue when that happened.

vatanaksoytezer, 2021-08-17:
Linking the relevant discourse post to track the next Galactic sync: https://discourse.ros.org/t/when-is-the-next-galactic-sync/21146/5.