moveit/moveit2 #608: Servo modifies the state of the PSM (planning scene monitor)

AndyZe, 2021-08-12:
`current_state_` is a pointer from the planning scene monitor:   

`current_state_ = planning_scene_monitor_->getStateMonitor()->getCurrentState();`

But, it gets modified here:

`current_state_->setJointGroupPositions(joint_model_group_, new_theta);`

I think that might cause some weirdness for anything else that uses the PSM.

henningkayser, 2021-08-12:
@AndyZe could you please link the code that you are referring to? `CurrentStateMonitor::getCurrentState()` actually returns a [copy](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L79), so internal state can be changed with the returned object. Maybe the documentation should be extended, or the function could be renamed to `copyCurrentState()` or similar...

AndyZe, 2021-08-12:
Here:  https://github.com/ros-planning/moveit2/blob/eda24029820019d630901062329a253732a7627d/moveit_ros/moveit_servo/src/servo_calcs.cpp#L723

Thanks, that is good to know. (The documentation could be improved...)

closed 2021-08-12 by AndyZe