moveit/moveit2 #624: Galactic build on buildfarm is failing since 2021-7-30

rhaschke, 2021-08-19:
... because package `controller_manager` is missing.
See https://build.ros2.org/job/Gdev__moveit__ubuntu_focal_amd64

vatanaksoytezer, 2021-08-19:
Great catch. We may want to fix and do a release before tomorrow, if we wanna catch the next sync: https://discourse.ros.org/t/preparing-for-galactic-sync-2021-08-24/21900

henningkayser, 2021-08-19:
It's actually because `ros2_control` has still not been released for Galactic (https://github.com/ros-controls/ros2_control/issues/421). On Rolling, everything is fine. We could technically remove all the run dependencies to `ros2_control` but this wouldn't really be a nice workaround, IMO.

rhaschke, 2021-08-19:
I see.

closed 2021-08-19 by rhaschke

gavanderhoorn, 2021-08-20:
Why not disable `source` builds on the buildfarm for now, until this has been resolved?

Getting spammed by tenths of "build for MoveIt on Galactic, etc, etc" per day is not very nice, especially not if we already know it's broken right now and we have our own CI here.