moveit/moveit2 #626: moveit_setup_assistant ros2 support

SalahSoliman, 2021-08-19:
I have checked all the branches for ros2 support for moveit_setup_assistant and I see that they're all depending on catkin, is it something I don't understand or it's still not updated?

vatanaksoytezer, 2021-08-19:
Hey @SalahSoliman, thanks for the issue! MSA is not ported to MoveIt 2 and ROS 2 yet. We are working on redesigning and porting it. See https://github.com/ros-planning/moveit2/issues/449 for some design discussions.

SalahSoliman, 2021-08-20:
@vatanaksoytezer thanks for you quick reply!
Do you have any suggestion for a work around until this part's done? also please lmk if I can help!

vatanaksoytezer, 2021-08-20:
@SalahSoliman your best quick bet would be just using the ROS 1 (moveit) MSA for now and edit the launch files that you need manually to fit MoveIt 2 and ROS 2.

vatanaksoytezer, 2021-08-24:
@SalahSoliman I am closing this issue for now but feel free to ask questions open discussions under the relevant issues or open a new one (or re-open this) if you need.

closed 2021-08-24 by vatanaksoytezer