moveit/moveit2 #630: Standardize moveit2 yaml parameter file format with ROS2 ros__parameter convention

shonigmann, 2021-08-19:
**Is your feature request related to a problem? Please describe.**
I noticed that in moveit2 and the ROS2 branches of moveit_resources, etc some of the yaml config files (e.g. [here](https://github.com/ros-planning/moveit2/blob/foxy/moveit_ros/moveit_servo/config/panda_simulated_config.yaml) or [here](https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/kinematics.yaml)) are still using the ROS1 format (parameter name: value), while others (e.g. [here](https://github.com/ros-planning/moveit2/blob/foxy/moveit_demo_nodes/run_ompl_constrained_planning/config/start_positions.yaml)) are using the ROS2 format (namespace -> node name -> ros__parameters --> parameter name / value). 

As a result, loading parameters in a node using a python launch file requires the use of [this ](https://github.com/ros-planning/moveit2/blob/292a99cb988710d04faab8e00412398fcfb986fd/moveit_ros/moveit_servo/launch/pose_tracking_example.launch.py#L21-L29) `load_yaml()` function, as opposed to the more "ROS2-esque" approach of simply including the path to the yaml file in question as a `Node` parameter. 

The latter would omit the need to define custom file loading functions in each launch file and would be more amenable to modularizing and generalizing launch files using `LaunchArgument`s to specify config files rather than having them as hard-coded paths. It would also potentially make things easier for users to use only the ROS2 format (at the cost of making it slightly harder to migrate ROS1 configurations).

**Describe the solution you'd like**
It would be great to standardize all yaml config files using the ROS2 format. That said, I'm fairly new to Moveit2 and am not sure if there would be more involved than just changing the config files themselves to include `ros__parameters`.

tylerjw, 2021-08-23:
One of the big problems is some of the moveit2 nodes can also be launched as component nodes.  In that case ros doesn't handle the ros__parameters prefix correctly.  If you look here: https://github.com/PickNikRobotics/parameter_tools/ @JafarAbdi has built something to standardize loading ros parameters for moveit.  Soon we plan on migrating the moveit launch files to use this tool.  Another thing we are doing to standardize parameters is I'm working on something to handle validating parameters in a more ros2 way.

shonigmann, 2021-09-08:
for the reference of anyone that come's across this issue, it looks like Component Nodes will soon support the same yaml file format as regular nodes (https://github.com/ros2/launch_ros/pull/259 ) but the parameter_tools package looks like a great solution as well.

sotanakamura, 2024-09-11:
I had a trouble due to ros__parameter when I loaded the yaml parameter files in a xml launch file or from command line arguments. As mentioned above, the problem of component nodes was solved, then we should make the yaml parameter files standard.