moveit/moveit2 #651: Discussion: Should we make CCov CI only run in push?

vatanaksoytezer, 2021-08-31:
CCov CI Jobs bottlenecks our CI time. Usually a nice cached CI job with Release builds takes around 8 to 20 minutes. But we still have to wait around an hour or so for CCov job to be completed (since it is built with RelWithDebugInfo).

For context example run from yesterday:
[image]

I propose we either run CCov with PR's that has a specific label, or only run it on push to save around 25 minutes every CI.

tylerjw, 2021-08-31:
Is there a way we could make it optional so it still runs with the PR but doesn't block the PR itself until it finishes?

vatanaksoytezer, 2021-08-31:
I think @henningkayser can do that

v4hn, 2021-08-31:
> Is there a way we could make it optional so it still runs with the PR but doesn't block the PR itself until it finishes?

Sure, it's a simple entry in the config for the branch. I'm not sure it's a good idea to remove it though. It's the only job testing `rolling-ci` and you probably want to test on rolling before merge.

Code coverage checks were originally proposed by @davetcoleman iirc.
Personally, I don't see the need to run this with each PR. I believe the original idea was to visualize how useful a PR is in terms of added tests. But  because we run with `RelWithDebInfo` even pull-requests that do *nothing* but add tests often are shown to *decrease* coverage.

> CCov CI Jobs bottlenecks our CI time

I'm not convinced 25 additional minutes are a bottleneck at all with MoveIt's development cycle. MoveIt2 lands a few more commits, but do you really feel that these 25 minutes limit your productivity? If you are concerned about reaching limits in GHA minutes, then it's probably a good idea to disable it altogether, but I would only take action here if we really observe paused jobs and can verify that's the reason.

--- Semi-related:
@vatanaksoytezer @rhaschke

Looking into the settings for the moveit2 repo, I noticed the clang-tidy job is currently allowed to fail - probably because it's based on `rolling-ci-testing` and it's not unlikely that `*-testing*` might fail sporadically for unrelated reasons.
Still, I would think `clang-tidy` should be enforced here as well. Not sure who currently manages the CI config for this repository.

vatanaksoytezer, 2021-08-31:
@v4hn thanks for chiming in and a detailed write-up!

> I'm not convinced 25 additional minutes are a bottleneck at all with MoveIt's development cycle. MoveIt2 lands a few more commits, but do you really feel that these 25 minutes limit your [...]

I don't think GHA minutes are a issue (or even limited AFAIK it's fully free for public repository). We usually (and understandably) wait for all CI jobs to complete  to approve / merge PRs. My motivation opening this issue is the question why should we wait for 45 minutes for a CI job to complete when we can do it 16 minutes. I don't think this would be huge benefit or change in our development time, but I feel it will "very slightly" improve our Merge / Review speed especially for trivial, small PRs that does not change core functionality (for eg. changing a doc, which can be merged in 15 minutes rather than in 45 minutes). I still don't think that's an issue, but why not?

> Looking into the settings for the moveit2 repo, I noticed the clang-tidy job is currently allowed to fail - probably because it's based on `rolling-ci-testing` and it's not unlikely that [...]

I think you have a good point here. I completely agree with you on enforcing clang-tidy. I will open a PR briefly to change clang-tidy job to galactic-ci (not rolling-ci because its already taking long, I want to make the CI jobs as parallel as possible) instead of rolling-ci-testing.

> Not sure who currently manages the CI config for this repository.

I think I am not the official person doing that (since I am no maintainer), but I really enjoy any DevOps stuff so trying to make PRs towards a solid and optimized CI jobs for moveit related parts of ros-planning org.

v4hn, 2021-08-31:
> I don't think GHA minutes are a issue

In that case I would propose to keep the job enforced.
Granted, I've also merged changes as admin that did not finish CI yet and "normal" maintainers can't do that.
But then again it is absolutely no problem to keep the tab open somewhere and just click merge before switching off the computer.

rhaschke, 2021-09-01:
I understand your motivation to have faster CI cycles. However, there is no problem defining the `ccov` job as optional in terms of a merge requirement (I just changed the config for `main`). Thus, any maintainer can merge a PR even when this job is not yet finished.
I suggest keeping the job as well because it facilitates requesting test coverage for new features from external contributors. 
In any case, we should understand/debug why the coverage build takes so much more time. It should profit from `ccache` as well, but obviously does not!

v4hn, 2021-09-01:
> I suggest keeping the job as well because it facilitates requesting test coverage for new features from external contributors. 

As I said above, it sadly does not - probably because of the `RelWithDebInfo` build.
https://github.com/ros-planning/moveit/pull/2665#issuecomment-840495632 is a nice illustration.

> In any case, we should understand/debug why the coverage build takes so much more time.
> It should profit from `ccache` as well, but obviously does not!

Possibly the problem is actually with the runtime of the tests and not with building.
MoveIt has at least [one test](https://github.com/ros-planning/moveit/blob/master/moveit_ros/robot_interaction/test/locked_robot_state_test.cpp) that can literally run forever when compiled in Debug mode. I never looked at its performance with `RelWithDebInfo`.
There might be other offenders too.

rhaschke, 2021-09-01:
Although release optimizations yield unreliable results, the reported results _do_ give a coarse indication of coverage.

> Possibly the problem is actually with the runtime of the tests and not with building.

Looking at the detailed timing or a recent CI job, the problem is with building (45min), not (yet) the tests (< 10min).

rhaschke, 2021-09-01:
Given that we migrated to GHA, having a 6h job timeout, we can consider switching back to a debug build.

v4hn, 2021-09-01:
> Given that we migrated to GHA, having a 6h job timeout, we can consider switching back to a debug build.

That would be great! We probably still want to make sure no single test runs unreasonably long in Debug mode...

Also, if the runtime becomes *much* longer by that change (>2h), I would actually propose to add a separate job for it that is not mandatory. It's important to compress conditions/tests into few CI jobs, but we still want noetic tests and a debug workspace build required for moveit and probably a rolling-ci test for moveit2.

vatanaksoytezer, 2021-09-01:
> That would be great! We probably still want to make sure no single test runs unreasonably long in Debug mode...

I can handle that, it is trivial to define timeouts in ros2 tests. By default ros2 tests already have a reasonable timeout of 60 seconds. I can change any tests as needed.

v4hn, 2021-09-01:
> > We probably still want to make sure no single test runs unreasonably long in Debug mode...
> 
> I can handle that, it is trivial to define timeouts in ros2 tests.

Of course I implied that the tests should be adapted to run in less time without loosing significant test conditions and not that they should run until timeout. ;)

> By default ros2 tests already have a reasonable timeout of 60 seconds.

I honestly don't know whether I consider a default timeout a good thing. After all execution speed can vary a lot across different systems.

vatanaksoytezer, 2021-09-08:
Closing in favor of #670.

closed 2021-09-08 by vatanaksoytezer

AndyZe, 2021-11-15:
I don't love CI runtimes of 2 hours.   :(

Personally I don't think ccov adds much. If somebody added a test or expanded a test, it should be pretty obvious that code coverage went up. I don't think we need the extra hour (or whatever it is) of CI runtime to tell us that.

It's not like we have so many tests they are starting to conflict.