moveit/moveit2 #655: install TARGETS given unknown argument "TARGETS"

cardboardcode, 2021-08-31:
### Description

Overview of your issue here.

### Your environment
* **ROS Distro**: Foxy
* **OS Version**: Ubuntu 20.04
* **Source** build
* Branch: [foxy](https://github.com/ros-planning/moveit2/tree/foxy)

### Steps to reproduce
**Reference**: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#readme

```bash
export COLCON_WS=~/workspace/ros_ws_foxy_ur_driver
mkdir -p $COLCON_WS/src
cd $COLCON_WS
git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos
rosdep install --ignore-src --from-paths src -y -r
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.bash
cd $COLCON_WS
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/MoveIt_Support.repos
vcs import src --skip-existing --input src/moveit2/moveit2.repos
rosdep install --ignore-src --from-paths src -y -r
cd $COLCON_WS/src/moveit2
git checkout foxy
cd $COLCON_WS/src/geometric_shapes
git checkout foxy
cd $COLCON_WS
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.bash
```

### Expected behaviour
`colcon build` command should build **moveit2** packages succesfully.

### Actual behaviour
`colcon build` command fails at building **moveit_core**.

### Backtrace or Console output
```bash
Starting >>> control_msgs
Starting >>> geometric_shapes
Starting >>> srdfdom
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> moveit_common
Starting >>> warehouse_ros
Starting >>> diffbot_description
Starting >>> ros2_control_test_nodes
Starting >>> rrbot_description
Starting >>> ur_client_library
Starting >>> ur_dashboard_msgs
Finished <<< ros2_control_test_assets [0.79s]
Starting >>> ur_msgs
Finished <<< rrbot_description [0.78s]
Starting >>> ur_description
Finished <<< diffbot_description [0.82s]
Starting >>> ur_moveit_config
Finished <<< ros2_control_test_nodes [1.00s]
Finished <<< moveit_common [1.14s]     
Finished <<< ur_description [0.34s]
Finished <<< ur_moveit_config [0.34s] 
Finished <<< ur_client_library [1.19s]
Finished <<< controller_manager_msgs [2.19s]
Finished <<< control_msgs [2.32s]     
Starting >>> hardware_interface
Finished <<< ur_msgs [1.62s]           
Finished <<< ur_dashboard_msgs [2.42s]
Finished <<< hardware_interface [0.48s]
Starting >>> controller_interface      
Starting >>> ros2_control_demo_hardware
Starting >>> transmission_interface
Finished <<< ros2_control_demo_hardware [0.35s]
Finished <<< transmission_interface [0.36s]
Finished <<< controller_interface [0.45s]
Starting >>> controller_manager
Finished <<< controller_manager [0.66s]
Starting >>> forward_command_controller
Starting >>> joint_trajectory_controller
Starting >>> joint_state_broadcaster
Starting >>> diff_drive_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> imu_sensor_broadcaster
Starting >>> ros2controlcli
Starting >>> ur_bringup
Starting >>> gripper_controllers
Finished <<< ur_bringup [0.92s]         
Finished <<< ros2controlcli [1.15s]     
Starting >>> ros2_control
Finished <<< joint_state_broadcaster [1.27s]
Finished <<< forward_command_controller [1.31s]
Finished <<< force_torque_sensor_broadcaster [1.27s]
Finished <<< imu_sensor_broadcaster [1.27s]
Starting >>> effort_controllers
Starting >>> position_controllers
Starting >>> velocity_controllers
Finished <<< diff_drive_controller [1.37s]
Finished <<< joint_trajectory_controller [1.40s]
Starting >>> ros2_control_demo_bringup
Starting >>> ur_controllers
Finished <<< gripper_controllers [1.38s]
Finished <<< ros2_control [0.48s]                                                                                                                                                                   
Finished <<< ros2_control_demo_bringup [0.38s]
Starting >>> ros2_control_demos
Finished <<< ur_controllers [0.63s]                                                                                                                                                                      
Starting >>> ur_robot_driver
Finished <<< velocity_controllers [0.73s]
Finished <<< effort_controllers [0.78s]                                                                                                                                 
Finished <<< position_controllers [0.77s]
Starting >>> ros2_controllers
Finished <<< ros2_control_demos [0.37s]
Finished <<< ros2_controllers [0.25s]                                                                                                                                                         
Finished <<< ur_robot_driver [0.42s]                                                                                                                             
Finished <<< srdfdom [14.0s]                                                                                                              
Finished <<< warehouse_ros [19.4s]                                                                            
Starting >>> warehouse_ros_mongo
Finished <<< warehouse_ros_mongo [13.2s]                                                                            
Finished <<< geometric_shapes [49.2s]                           
Starting >>> moveit_core
--- stderr: moveit_core                            
CMake Error at collision_detection_bullet/CMakeLists.txt:51 (install):
  install TARGETS given unknown argument "TARGETS".

---
Failed   <<< moveit_core [9.53s, exited with code 1]

Summary: 39 packages finished [59.0s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core
  21 packages not processed
```

JafarAbdi, 2021-08-31:
@cardboardcode Could you please test this https://github.com/ros-planning/moveit2/pull/658 .?

cardboardcode, 2021-08-31:
Testing now. Will update results here in this comment.

cardboardcode, 2021-08-31:
## Solution (Crude Workaround) :rocket:

For those who are facing this issue, run the commands below to resolve this issue:

```bash
export COLCON_WS=~/workspace/ros_ws_foxy_ur_driver
cd $COLCON_WS/src/moveit2
git checkout JafarAbdi-patch-1
cd $COLCON_WS
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

## Verify that it works
If the `colcon build` command builds the packages without failure, it worked.

closed 2021-08-31 by cardboardcode